gem5/gem5 #15: Potential segmentation fault in /src/cpu/pred/ltage.cc:556

HanFa, 2018-02-15:
In the update function when trying to allocate new entries, is it possible that changing the value of X (line 556 558) might result in potential segfault in (line 562)? To me it seems that there is no any bound checking.

```c++
            int Y = nrand &
                ((ULL(1) << (nHistoryTables - bi->hitBank - 1)) - 1);
            int X = bi->hitBank + 1;
            if (Y & 1) {
                X++;
                if (Y & 2)
                    X++;
            }
            // No entry available, forces one to be available
            if (min > 0) {
                gtable[X][bi->tableIndices[X]].u = 0;
            }
```

yangych29, 2018-02-15:
I met some segfaults when I run the simulator with Ltage branch predictor. I think here are some potential bugs.

```
./perlbench.sh: line 11:  7124 Segmentation fault      (core dumped) ../build/ARM/gem5.opt --outdir=$OUTPUTDIR ../configs/example/se.py --cmd=$CMD -o "$TRAIN_OPT" --cpu-type=O3_ARM_v7a_3 --caches --branch-pred=$PRED
```

powerjg, 2018-04-13:
I am currently closing all issues on github and disabling the issue support.

If you have general gem5 questions, please use the gem5-users mailing list (http://www.gem5.org/mailman/listinfo/gem5-users).

Please report any issues/bugs to the gem5-dev mailing list (http://www.gem5.org/mailman/listinfo/gem5-dev). However, gem5 is an open source project, so you are encoraged to fix any bugs/issues you find an submit a review request on our review site: https://gem5-review.googlesource.com

More details can be found in the CONTRIBUTING file: https://gem5.googlesource.com/public/gem5/+/master/CONTRIBUTING.md

closed 2018-04-13 by powerjg