gem5/gem5 #16: Linux kernel boot on x86 full system (FS) with --cpu-type=DerivO3CPU fails with: "Kernel panic - not syncing: Attempted to kill the idle task!"

cirosantilli-work, 2018-02-27:
fbe63074e3a8128bdbe1a5e8f6509c565a3abbd4

Boot works if I remove `--cpu-type=DerivO3CPU`.

Possible same on mailing list: https://www.mail-archive.com/gem5-users@gem5.org/msg14062.html but `DeriveO3CPU` not mentioned so I'm not sure.

Automated setup to reproduce: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/cc347bbef76687f71e19a5e75f8a92f9e8ad6332#gem5-change-system-parameters

Full gem5 command:

```
M5_PATH='/data/git/linux-kernel-module-cheat/gem5/gem5-system' '/data/git/linux-kernel-module-cheat/gem5/gem5/build/X86/gem5.opt'  '/data/git/linux-kernel-module-cheat/gem5/gem5/configs/example/fs.py' --command-l
ine='earlyprintk=ttyS0 console=ttyS0 lpj=7999923 root=/dev/hda nokaslr norandmaps printk.devkmsg=on printk.time=y' --disk-image='/data/git/linux-kernel-module-cheat/buildroot/output.x86_64-gem5~/images/rootfs.ext
2' --kernel='/data/git/linux-kernel-module-cheat/buildroot/output.x86_64-gem5~/build/linux-custom/vmlinux' --num-cpus=1  --caches --l1d_size=1024 --l2cache --l2_size=1024 --l3_size=1024 --cpu-type=DerivO3CPU
```

dmesg excerpt:

```
    [    0.010765] BUG: unable to handle kernel NULL pointer dereference at 000000000000022e                 
    [    0.010825] IP: text_poke_early+0x1f/0x27        
    [    0.010855] PGD 0 P4D 0                          
    [    0.010879] Oops: 0002 [#1] SMP PTI              
    [    0.010905] Modules linked in:                   
    [    0.010931] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.15.0 #1                                       
    [    0.010971] Hardware name:  , BIOS  06/08/2008   
    [    0.011000] RIP: 0010:text_poke_early+0x1f/0x27  
    [    0.011000] RSP: 0000:ffffffff81c03dd0 EFLAGS: 000008b8                                               
    [    0.011000] RAX: 0000000000000291 RBX: ffffffff8200a228 RCX: 0000000000000000                         
    [    0.011000] RDX: 000000000000026c RSI: ffffffff81c03de9 RDI: 000000000000026c                         
    [    0.011000] RBP: ffffffff81c03de2 R08: ffffffff810145eb R09: 0000000000000005                         
    [    0.011000] R10: ffffffff81c03de2 R11: ffffffff81c03de2 R12: ffffffff820154c4                         
    [    0.011000] R13: ffffffff81fab2e0 R14: 0000000000000000 R15: 0000000000000000                         
    [    0.011000] FS:  0000000000000000(0000) GS:ffff88001fc00000(0000) knlGS:0000000000000000              
    [    0.011000] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033                                         
    [    0.011000] CR2: 000000000000022e CR3: 0000000001c0a000 CR4: 00000000000006b0                         
    [    0.011000] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000                         
    [    0.011000] DR3: 0000000000000000 DR6: 00000000ffff0ff0 DR7: 0000000000000000                         
    [    0.011000] Call Trace:                          
    [    0.011000]  ? apply_paravirt.part.1+0x74/0x82   
    [    0.011000]  ? vprintk_emit+0x201/0x213          
    [    0.011000]  ? printk+0x3e/0x46                  
    [    0.011000]  ? free_reserved_area+0x103/0x112    
    [    0.011000]  ? alternative_instructions+0xbf/0xcf
    [    0.011000]  ? check_bugs+0x1be/0x2f2            
    [    0.011000]  ? start_kernel+0x4ac/0x4c3          
    [    0.011000]  ? secondary_startup_64+0xa5/0xb0    
    [    0.011000] Code: 30 eb ba 31 c0 c3 b8 01 00 00 00 c3 49 89 f8 48 89 d1 9c 58 66 66 90 66 90 48 89 c2 fa 66 66 90 66 66 90 f3 a4 48 89 d7 57 9d 66 <66> 90 66 90 4c 89 c0 c3 41 57 41 56 41 55 41 54 55 53 48 89 fb                                                 
    [    0.011000] RIP: text_poke_early+0x1f/0x27 RSP: ffffffff81c03dd0                                      
    [    0.011000] CR2: 000000000000022e                
    [    0.011000] ---[ end trace 6d53c1c17a70e53c ]--- 
    [    0.011000] Kernel panic - not syncing: Attempted to kill the idle task!                              
    [    0.011000] ---[ end Kernel panic - not syncing: Attempted to kill the idle task!
```

hoangt, 2018-03-22:
With --cpu-type=DerivO3CPU, I could not boot full system for X86. Same to you, removing that option then everything is fine.

powerjg, 2018-04-13:
I am currently closing all issues on github and disabling the issue support.

If you have general gem5 questions, please use the gem5-users mailing list (http://www.gem5.org/mailman/listinfo/gem5-users).

Please report any issues/bugs to the gem5-dev mailing list (http://www.gem5.org/mailman/listinfo/gem5-dev). However, gem5 is an open source project, so you are encoraged to fix any bugs/issues you find an submit a review request on our review site: https://gem5-review.googlesource.com

More details can be found in the CONTRIBUTING file: https://gem5.googlesource.com/public/gem5/+/master/CONTRIBUTING.md

closed 2018-04-13 by powerjg