gem5/gem5 #17: repetition problem of (begin simulation ..... and simulation) in the stats.txt file

HADOUDA, 2018-03-08:
please, help me, when i have run the benchmark parsec i have some repetition of (begin simulation ...... end simulation) in the stats.txt file with some informations defferent and i'm looking for the execution time .

cirosantilli-work, 2018-03-09:
Hi HADOUDA,

In order to make it more likely that you will get your problem solved, please provide more details:

- what is your gem5 version
- what command did you run
- what was the relevant stdout output and terminal output
- what did you expect to see instead

HADOUDA, 2018-03-09:
hi,
-I uploaded the latest version of gem5 in gihub, I executed the following command (./build/X86/gem5.opt configs/example/fs.py --kernel=/media/ali/ali/gem5/full_system/binaries/x86_64-vmlinux-2.6.22.9.smp)with disk image x86root-parsec.img.
i used a .rcs script to run a parsec application eg blackscholes, when i saw the stats file .txt i found several parts of (begin simulation ............ ..and simulation) with deferent information for example the execution time; because I think that in reality I found  one part of (begin simulation .... end simulation)
to take the correct execution time .
thank you;

cirosantilli-work, 2018-03-13:
OK, I think I understand now.

It is likely that the benchmark you are trying to run is already instrumented with magic instructions that make gem5 dump the stats, just like `m5 dumpstats` does.

More details can be found at: https://www.mail-archive.com/gem5-users@gem5.org/msg15159.html

closed 2018-03-15 by HADOUDA