gem5/gem5 #18: xx86root-parsec.img image with the kernel x86_64-vmlinux-2.6.28.4-smp problem

HADOUDA, 2018-03-08:
hi,please help me ,I have a problem, when I executed the x86root-parsec.img image with the kernel x86_64-vmlinux-2.6.28.4-smp I find this:
<ffffffff8021ef40>] stop_this_cpu+0x0/0x30
 [<ffffffff8025db9e>] smp_call_function_mask+0x1de/0x250
 [<ffffffff803a3f78>] vsnprintf+0x268/0x6b0
 [<ffffffff80615074>] printk+0x40/0x45
 [<ffffffff8021ef30>] native_smp_send_stop+0x20/0x30
 [<ffffffff80614f8d>] panic+0x82/0x129
 [<ffffffff8023ecae>] do_exit+0x7de/0x890
 [<ffffffff80615074>] printk+0x40/0x45
 [<ffffffff806184ba>] oops_end+0x7a/0xc0
 [<ffffffff8020d3d4>] do_invalid_op+0x84/0xa0
 [<ffffffff808d69e7>] xsave_cntxt_init+0x35/0x130
 [<ffffffff8023b6de>] release_console_sem+0x1ae/0x200
 [<ffffffff802547dd>] up+0xd/0x40
 [<ffffffff806178e9>] error_exit+0x0/0x51
 [<ffffffff808d69e7>] xsave_cntxt_init+0x35/0x130
 [<ffffffff808d69e7>] xsave_cntxt_init+0x35/0x130
 [<ffffffff8060e2b1>] fpu_init+0x4a/0x97
 [<ffffffff8060fa3f>] cpu_init+0x319/0x33f
 [<ffffffff808cdab5>] start_kernel+0x1b2/0x321
 [<ffffffff808cd405>] x86_64_start_kernel+0xd9/0xdd
---[ end trace 4eaa2a86a8e2da22 ]---

cirosantilli-work, 2018-03-09:
Hello HADOUDA,

I don't know about the img you are using, but this may be another good way to run PARSEC on gem5: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/03b1a720c47352a7b8d0a13eb325f7a079b953c9#parsec-benchmark

powerjg, 2018-04-13:
I am currently closing all issues on github and disabling the issue support.

If you have general gem5 questions, please use the gem5-users mailing list (http://www.gem5.org/mailman/listinfo/gem5-users).

Please report any issues/bugs to the gem5-dev mailing list (http://www.gem5.org/mailman/listinfo/gem5-dev). However, gem5 is an open source project, so you are encoraged to fix any bugs/issues you find an submit a review request on our review site: https://gem5-review.googlesource.com

More details can be found in the CONTRIBUTING file: https://gem5.googlesource.com/public/gem5/+/master/CONTRIBUTING.md

closed 2018-04-13 by powerjg