gem5/gem5 #21: Linux kernel booted on x86 Full System mode

XX124188, 2018-03-24:
Hi, I`m trying to boot x86 gem5 full system simulation with the latest Linux kernel version 4.x 
but didn`t work. 
I built kernel version 4.12.3,4.15.12 and 3.11 for testing, and build with default config : `make defconfig`
can someone share your kernel configuration setup that played well on Gem5 ?

thanks.

cirosantilli-work, 2018-03-24:
@XX124188 Linux v4.15 booted for me with this configuration: https://github.com/cirosantilli/linux-kernel-module-cheat/blob/29908dffb2726b0819fdf95240ae513dddb8af48/kernel_config_x86_64-gem5 and with this setup: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/29908dffb2726b0819fdf95240ae513dddb8af48#gem5-getting-started

That config was originally taken from: http://web.archive.org/web/20171229121642/http://www.lowepower.com/jason/files/config

XX124188, 2018-03-24:
@cirosantilli-work I used the config file above to build kernel 4.15.12 and set the rest of kernel configuration by default. It seems to be working fine now.
Thank you for your help !!

cirosantilli-work, 2018-03-25:
Awesome. Let's close this issue if it is solved.

closed 2018-03-26 by XX124188