gem5/gem5 #23: Error Compiling with LLVM 6.0 on Ubuntu Artful

lvcargnini, 2018-03-26:
Hello, 
I'm trying to build the X86_MOESI_AMD_Base model of gem5 using clang/llvm. I got it running las week with gcc-7 on Ubuntu artful. I have it running on RHEL 6, using clang 5. 
So, trying to build it using the llvm distributed by LLVM.org, I'm gettign the following error on uBuntu Artful:
```
7 warnings generated.
 [     CXX] X86_MOESI_AMD_Base/mem/ruby/common/Address.cc -> .o
In file included from build/X86_MOESI_AMD_Base/mem/ruby/common/Address.cc:29:
In file included from build/X86_MOESI_AMD_Base/mem/ruby/common/Address.hh:37:
In file included from build/X86_MOESI_AMD_Base/base/types.hh:43:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/memory:81:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr.h:52:
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:588:8: error: delete called on non-final 'Stats::ScalarStatNode' that has virtual functions but non-virtual
      destructor [-Werror,-Wdelete-non-virtual-dtor]
              delete __p;
              ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:595:4: note: in instantiation of function template specialization
      'std::__shared_count<__gnu_cxx::_S_atomic>::__shared_count<Stats::ScalarStatNode *>' requested here
        : __shared_count(__p)
          ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:1079:17: note: in instantiation of function template specialization
      'std::__shared_count<__gnu_cxx::_S_atomic>::__shared_count<Stats::ScalarStatNode *>' requested here
        : _M_ptr(__p), _M_refcount(__p, typename is_array<_Tp>::type())
                       ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr.h:129:25: note: in instantiation of function template specialization 'std::__shared_ptr<Stats::Node,
      __gnu_cxx::_S_atomic>::__shared_ptr<Stats::ScalarStatNode, void>' requested here
        shared_ptr(_Yp* __p) : __shared_ptr<_Tp>(__p) { }
                               ^
build/X86_MOESI_AMD_Base/base/statistics.hh:3030:11: note: in instantiation of function template specialization 'std::shared_ptr<Stats::Node>::shared_ptr<Stats::ScalarStatNode, void>' requested here
        : node(new ScalarStatNode(s.info()))
          ^
In file included from build/X86_MOESI_AMD_Base/mem/ruby/common/Address.cc:29:
In file included from build/X86_MOESI_AMD_Base/mem/ruby/common/Address.hh:37:
In file included from build/X86_MOESI_AMD_Base/base/types.hh:43:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/memory:81:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr.h:52:
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:376:9: error: delete called on non-final 'Stats::ScalarStatNode' that has virtual functions but non-virtual
      destructor [-Werror,-Wdelete-non-virtual-dtor]
      { delete _M_ptr; }
        ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:371:7: note: in instantiation of member function 'std::_Sp_counted_ptr<Stats::ScalarStatNode *,
      __gnu_cxx::_S_atomic>::_M_dispose' requested here
      _Sp_counted_ptr(_Ptr __p) noexcept
      ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:584:20: note: in instantiation of member function 'std::_Sp_counted_ptr<Stats::ScalarStatNode *,
      __gnu_cxx::_S_atomic>::_Sp_counted_ptr' requested here
              _M_pi = new _Sp_counted_ptr<_Ptr, _Lp>(__p);
                          ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:595:4: note: in instantiation of function template specialization
      'std::__shared_count<__gnu_cxx::_S_atomic>::__shared_count<Stats::ScalarStatNode *>' requested here
        : __shared_count(__p)
          ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:1079:17: note: in instantiation of function template specialization
      'std::__shared_count<__gnu_cxx::_S_atomic>::__shared_count<Stats::ScalarStatNode *>' requested here
        : _M_ptr(__p), _M_refcount(__p, typename is_array<_Tp>::type())
                       ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr.h:129:25: note: in instantiation of function template specialization 'std::__shared_ptr<Stats::Node,
      __gnu_cxx::_S_atomic>::__shared_ptr<Stats::ScalarStatNode, void>' requested here
        shared_ptr(_Yp* __p) : __shared_ptr<_Tp>(__p) { }
                               ^
build/X86_MOESI_AMD_Base/base/statistics.hh:3030:11: note: in instantiation of function template specialization 'std::shared_ptr<Stats::Node>::shared_ptr<Stats::ScalarStatNode, void>' requested here
        : node(new ScalarStatNode(s.info()))
          ^
In file included from build/X86_MOESI_AMD_Base/mem/ruby/common/Address.cc:29:
In file included from build/X86_MOESI_AMD_Base/mem/ruby/common/Address.hh:37:
In file included from build/X86_MOESI_AMD_Base/base/types.hh:43:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/memory:81:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr.h:52:
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:588:8: error: delete called on non-final 'Stats::VectorStatNode' that has virtual functions but non-virtual
      destructor [-Werror,-Wdelete-non-virtual-dtor]
              delete __p;
[... 448 lines cut ...]
              ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:595:4: note: in instantiation of function template specialization
      'std::__shared_count<__gnu_cxx::_S_atomic>::__shared_count<Stats::ConstNode<long> *>' requested here
        : __shared_count(__p)
          ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr_base.h:1079:17: note: in instantiation of function template specialization
      'std::__shared_count<__gnu_cxx::_S_atomic>::__shared_count<Stats::ConstNode<long> *>' requested here
        : _M_ptr(__p), _M_refcount(__p, typename is_array<_Tp>::type())
                       ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/7.2.0/../../../../include/c++/7.2.0/bits/shared_ptr.h:129:25: note: in instantiation of function template specialization 'std::__shared_ptr<Stats::Node,
      __gnu_cxx::_S_atomic>::__shared_ptr<Stats::ConstNode<long>, void>' requested here
        shared_ptr(_Yp* __p) : __shared_ptr<_Tp>(__p) { }
                               ^
build/X86_MOESI_AMD_Base/base/statistics.hh:3130:11: note: in instantiation of function template specialization 'std::shared_ptr<Stats::Node>::shared_ptr<Stats::ConstNode<long>, void>' requested here
        : node(new ConstNode<signed long>(value))
          ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
scons: *** [build/X86_MOESI_AMD_Base/mem/ruby/common/Address.o] Error 1
scons: building terminated because of errors.

```
I only observe this error using clang. 

I use LLVM because the final binaries are in some way better optimized, and they have a better performance compared to the gcc ones.

lvcargnini, 2018-03-26:
I forgot to mention it is the same for all architectures ARM, RISCV, X86, I just used x86, because was my last attempt.

lvcargnini, 2018-03-27:
it seems that on LLVM 5.x it is compiling without any issues so far, so it seems relative to LLVM 6.x

cirosantilli-work, 2018-04-02:
We should fix this, but just curious, why do you want to build with LLVM?

powerjg, 2018-04-13:
I am currently closing all issues on github and disabling the issue support.

If you have general gem5 questions, please use the gem5-users mailing list (http://www.gem5.org/mailman/listinfo/gem5-users).

Please report any issues/bugs to the gem5-dev mailing list (http://www.gem5.org/mailman/listinfo/gem5-dev). However, gem5 is an open source project, so you are encoraged to fix any bugs/issues you find an submit a review request on our review site: https://gem5-review.googlesource.com

More details can be found in the CONTRIBUTING file: https://gem5.googlesource.com/public/gem5/+/master/CONTRIBUTING.md

closed 2018-04-13 by powerjg