gem5/gem5 #25: Gem5 simulation speed too slow

XX124188, 2018-04-03:
Hi, I`m running ubuntu server on gem5 now, and the simulation speed is too slow that it takes couple minutes to complete only one command. Is that normal?

Running gem5 :  scons build/x86/gem5.opt  config/example/fs.py --kernel=vmlinux --disk-image=x86.img --mem-size=8G

I`ve checked cpu type in config.ini under m5out/, and shows type=AtomicSimpleCPU.

Does anyone know how to speed up gem5 simulation?

cirosantilli-work, 2018-04-03:
What command takes a couple of minutes? gem5.opt itself to boot the Linux kernel, or some other command inside the guest?

A few minutes for a kernel boot is normal depending on your machine / kernel config, see these benchmarks: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/f3503b4cc810556df3c736d0a147cc54e05efc83#gem5-vs-qemu-performance

This is basically why `m5 checkpoint` is so important.

cirosantilli-work, 2018-04-08:
> Does gem5 reads the boot args on disk image or take its own setting in gem5 dir?

`fs.py` has `--command-line` for kernel CLI args, have a look at `gem5.opt fs.py --help`.

closed 2018-04-10 by XX124188