gem5/gem5 #26: "end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)" on fs_bigLITTLE.py

cirosantilli-work, 2018-04-08:
I've been asked by email:

```
./build/ARM/gem5.opt configs/example/arm/fs_bigLITTLE.py --caches --kernel=/media/ali/ali/gem5/linux-arm64-gem5-gem5/vmlinux --dtb=/media/ali/ali/gem5/system/arm/dt/armv8_gem5_v1_big_little_2_2.dtb --disk=/media/ali/ali/gem5/fs_armv8/disks/aarch64-ubuntu-trusty-headless.img --bootscript=/media/ali/ali/gem5/system/arm/aarch64_bootloader/boot_emm.arm64
```

failed with:

```
end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```

cirosantilli-work, 2018-04-08:
I have the following advice:

A bit above the kernel message you showed, there should be a message of type:

```
[    2.139471] Please append a correct "root=" boot option; here are the available partitions:
[    2.140002] fe00          524288 vda
```

This shows which disks you can pass as a kernel command line option with `root=`.

If the list is empty, it could mean that the kernel does not have a required driver to read the disk image format, your best bet is to copy the options from the original Ubuntu kernel.

Finally, maybe first try to boot with QEMU, which boots much faster than gem5.

I'm closing the issue because I'm pretty certain it is not a gem5 bug, but we can continue the conversation here nonetheless.

cirosantilli-work, 2018-04-08:
Ah, OK, let's reopen this then for now.

cirosantilli-work, 2018-04-08:
I would try to understand how to pass kernel command line paramters to `fs_bigLITTLE.py`, since we need `root=`, but there is no `--command-line` argument as in `fs.py`.

HADOUDA, 2018-04-08:
here, for example, how to use arguments because it's not like the fs.py mode;

usage: fs_bigLITTLE.py [-h] [--restore-from RESTORE_FROM] [--dtb DTB]
                        [--kernel KERNEL] [--disk DISK]
                        [--bootscript BOOTSCRIPT]
                        [--cpu-type {timing, atomic, exynos}]
                        [--kernel-init KERNEL_INIT] [--big-cpus BIG_CPUS]
                        [--little-cpus LITTLE_CPUS] [--caches]
                        [--last-cache-level LAST_CACHE_LEVEL]
                        [--big-cpu-clock BIG_CPU_CLOCK]
                        [--little-cpu-clock LITTLE_CPU_CLOCK]
                        [--sim-quantum SIM_QUANTUM]
thanks

powerjg, 2018-04-13:
I am currently closing all issues on github and disabling the issue support.

If you have general gem5 questions, please use the gem5-users mailing list (http://www.gem5.org/mailman/listinfo/gem5-users).

Please report any issues/bugs to the gem5-dev mailing list (http://www.gem5.org/mailman/listinfo/gem5-dev). However, gem5 is an open source project, so you are encoraged to fix any bugs/issues you find an submit a review request on our review site: https://gem5-review.googlesource.com

More details can be found in the CONTRIBUTING file: https://gem5.googlesource.com/public/gem5/+/master/CONTRIBUTING.md

closed 2018-04-13 by powerjg