gem5/gem5 #27: How to enable network in the X86 full system simulation

lwj0012, 2018-04-11:
Is this function supported? Or how can I accomplish this?
Someone has given me a patch manual (which can be found [X86 networking(1).pdf](https://github.com/gem5/gem5/files/1896825/X86.networking.1.pdf)
) to follow, but both of us don't know which version of Gem5 can apply for the patch.
Any advice? Thanks.

cirosantilli-work, 2018-04-11:
Related: https://stackoverflow.com/questions/48941494/how-to-do-port-forwarding-from-guest-to-host-and-vice-versa-in-gem5

lwj0012, 2018-04-11:
@cirosantilli-work Yes, the link you gave is somehow related to my issue, which both need the network interface in X86 mode. But there is still no answer, or should I take the "it is impossible" answer?

cirosantilli-work, 2018-04-11:
Not possible seems likely now, given Jason's answer: https://www.mail-archive.com/gem5-users@gem5.org/msg15209.html (linked to from the SO post).

But I feel there is more investigation to be done.

powerjg, 2018-04-13:
I am currently closing all issues on github and disabling the issue support.

If you have general gem5 questions, please use the gem5-users mailing list (http://www.gem5.org/mailman/listinfo/gem5-users).

Please report any issues/bugs to the gem5-dev mailing list (http://www.gem5.org/mailman/listinfo/gem5-dev). However, gem5 is an open source project, so you are encoraged to fix any bugs/issues you find an submit a review request on our review site: https://gem5-review.googlesource.com

More details can be found in the CONTRIBUTING file: https://gem5.googlesource.com/public/gem5/+/master/CONTRIBUTING.md

closed 2018-04-13 by powerjg