gem5/gem5 #61: Can't generate compile_commands.json by using scons compilation_db

jlau002, 2023-06-21:
I want to generate a compilation database by using compilation_db in scons.
I add the following code to SConstruct file, but it don't work
`main.Tool('compilation_db')
main.CompilationDatabase('compile_commands.json')`

I add .Tool('compilation_db') and .CompilationDatabase('compile_commands.json') to many env on SConstruct and src/SConscript, it don't work too.

It's confusing to me. How to generate compilation database of gem5?
I'm using ubuntu 22.04, the scons version is 4.0.1

powerjg, 2023-06-22:
This is a good question. I've never used the compilation DB from scons. 

We have a lot of very esoteric scons extensions in gem5, so it's possible that scons tools that have not been specialized to work with gem5 won't work out of the box. On the other hand, there has been some work to make scons more compatible with IDEs.

Two suggestions:
1. If you have some specific error messages, we may be able to help.
2. You can look at the `site_scons` directory to see if there's something there that could be helpful

powerjg, 2023-07-11:
Closing since no response in 3 weeks. Feel free to reopen to continue the conversation.

closed 2023-07-11 by powerjg

ipocentro87, 2025-03-09:
Hi all, I have been trying to generate `compile_commands.json` as well, with `compilation_db`, without success.
Does anyone know how to make it work?

powerjg, 2025-03-09:
Does this work for you? https://github.com/gem5/gem5/pull/1942

ipocentro87, 2025-03-10:
Hi @powerjg it seems to work indeed, I am going to test it for a couple of days to see whether it really works.
I was thinking that's perhaps a good idea to merge the change.

powerjg, 2025-03-10:
Great! Can you review #1942 once you've used it for a few days? I'll merge after that.