gem5/gem5 #62: "SyntaxError: Can't build lexer" on Ubuntu 23.04

alexredd99, 2023-07-08:
I am getting the following error when trying to compile gem5 on Ubuntu 23.04 with scons 4.0.1:

`ERROR:fm_proj_ply:/home/alexander/Desktop/junkyard/ramreuse/dev/gem5/build/X86/arch/arm/fastmodel/SConscript:220: Invalid regular expression for rule 't_STRLIT'. global flags not at the start of the expression at position 13
SyntaxError: Can't build lexer`

I've been able to build gem5 in the gem5 Ubuntu 22.04 Docker image but I'm not sure why I'm getting this error when compiling on my own machine. Thank you for the help.

powerjg, 2023-07-11:
What version of gem5 are you using? I believe that gem5 23.0 (released over the weekend) fixes this issue. Can you try pulling the stable branch and checking if this is still a problem?

Note: The underlying problem (I think) was with Python 3.11 and ply.

BobbyRBruce, 2023-07-18:
I'm pretty sure this bug is due to gem5 <v23 being incompative with Python 3.11. Given the lack of of response I'm closing this issue. 

Those with this issue should obtain the latest version of gem5 (or at least >=v23).

closed 2023-07-18 by BobbyRBruce

alexredd99, 2023-07-18:
Apologies for the late response, yes the latest release did fix my problem! Thank you for the help.