gem5/gem5 #74: New HashHelper tests not compiling with gcc-7.5.0

giactra, 2023-07-13:
I am encountering a compilation problem with the new HashHelper unit-tests [1]:

`unordered_map.h:133:55: error: 'value' is not a member of 'std::__not_<std::__and_<std::__is_fast_hash<std::hash<std::tuple<int, bool, int**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (*)(float)> > >, std::__detail::__is_noexcept_hash<std::tuple<int, bool, int**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (*)(float)>, std::hash<std::tuple<int, bool, int**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (*)(float)> > > > >'
       using node_type = typename _Hashtable::node_type;`

The problem doesn't seem to be present in newer versions of gcc so I am just changing the version. However we do support gcc-7 in gem5 so either we work around the problem or we deprecate gcc-7 support @BobbyRBruce @powerjg.
gcc-7 is default compiler for Ubuntu18.04 which is past its EOL anyway so I believe it is a reasonable solution

[1]: https://gem5-review.googlesource.com/c/public/gem5/+/67663

giactra, 2023-07-13:
@MeatBoy106

MeatBoy106, 2023-07-13:
Can you attach the complete error message please? I don't have easy access to gcc 7.

Based on that message, it looks like a libstdc++ bug as I would expect a user meta programming error to show up more gracefully, i.e., not a substitution failure on an implementation class template like std::__not_.

In all cases, I suggest disabling hash helpers for gcc 7 until definitive removal of gcc 7 support.

giactra, 2023-07-13:
> Can you attach the complete error message please? I don't have easy access to gcc 7.
> 

I will, but FYI you can probably get one by pulling the 18.04 image:

`docker pull gcr.io/gem5-test/ubuntu-18.04_all-dependencies:v22-1`

> Based on that message, it looks like a libstdc++ bug as I would expect a user meta programming error to show up more gracefully, i.e., not a substitution failure on an implementation class template [...]

That would mean an object relying on them will have to implement 2 behaviours depending on the compiler version.. I have to say I am not a huge fan of this but I am open to hear other people's opinion

giactra, 2023-07-13:
Edit: reading the comment more carefully, I realise this would be a temporary solution

powerjg, 2023-07-13:
Here's what we "support" and test: https://github.com/gem5/gem5/blob/stable/.github/workflows/compiler-tests.yaml#L19

I would be OK with dropping support for 18.04 and GCC < 8. We could also drop support for clang < 7 if we are going to call 2018 the cut off. Now would be a good time to do this since we just released gem5.

BobbyRBruce, 2023-07-14:
#80 Drops support for GCC 7 and Clang 6.0, as well as Ubuntu 18.04. Currently a draft as I want to check a few things before merging.