gem5/gem5 #89: Remove "python/m5/info.py"

BobbyRBruce, 2023-07-14:
This file is automatically generated during compilation and contains a copy of the LICENSE, COPYING, and README.md, which is then compiled into the gem5 binary.

As pointed out here: https://github.com/gem5/gem5/pull/71#discussion_r1263935941, this is not needed and should probably be removed to avoid complexity.

andysan, 2023-07-18:
Including licenses is pretty standard and something we _might_ want to keep. The current implementation isn't that useful though since it doesn't aggregate all of the various licenses (e.g., from ext) covering gem5. On the other hand, I can't think of any use case where anyone would be shipping gem5 binaries, so it might not be very relevant.

BobbyRBruce, 2023-07-18:
> Including licenses is pretty standard and something we _might_ want to keep. The current implementation isn't that useful though since it doesn't aggregate all of the various licenses (e.g., from [...]

Good point.

The `--copyright` argument, `gem5.opt --copyright`,  currently provides users with info on gem5 licensing/copyright which includes mention of the "ext" licenses and the slight variances in license in gem5. I'm not sure if this is sufficient to count as a license, but perhaps it could be expanded to do so?

Sidenote: The `--copyright` argument text appears to have been untouched in some time, the stated copyrights don't appear to have been updated in over 5 years.

powerjg, 2023-07-18:
Displaying a single line or two for the license makes sense. I don't know of many tools (I can't find any on my Linux box with a cursory look) which have a `--copyright` option that displays the entire license text. I don't think that we need a `--copyright` option.

Of course, if anyone distributed a gem5 binary they would also need to distribute the copyright files. But that would be up to them. I think it's odd to embed that in the binary.

andysan, 2023-07-19:
It might be less common in command line tools, but GUI applications typically do include full licenses. For example, Firefox uses the magic URL about:license to show all of the licenses covering a specific Firefox build. I'm pretty sure I have seen that in other GUI tools like KiCad and FreeCAD as well.

BobbyRBruce, 2023-08-17:
Closing due to lack of agreement on how best to do this. Not an important change.

closed 2023-08-17 by BobbyRBruce