gem5/gem5 #94: macOS build failure due to conflicting defines in gdbremote/signals.hh

austinharris, 2023-07-15:
This commit breaks building on macOS https://github.com/gem5/gem5/commit/7230a3e7f0f23621d9d09df3f7420c08a08cc118 as EXC_* are already defined in exception_types.h. I'm happy to submit a PR undef'ing these, but I'm not certain if that is the desired fix?

powerjg, 2023-07-17:
I'm a bit surprised this hasn't been caught as we often test on MacOS. Could you send your exact host setup (version of MacOS, etc.) and the exact build command you're using? 

The solution is probably guarding these with `#ifdef macos` or something like that (I forget the exact syntax).

austinharris, 2023-07-17:
macOS Ventura 13.4.1. I am building with scons build/ALL/gem5.opt with clang 11.1.0. The error arises from libSystem 11.0.0:

`libSystem-11.0.0/include/mach/exception_types.h:87:33: note: expanded from macro 'EXC_BREAKPOINT'
#define EXC_BREAKPOINT          6       /* Trace, breakpoint, etc. */`

I don't think guarding these with ifdef is the solution as the values don't match. Let me know if there is any other info on dependencies needed.

BobbyRBruce, 2023-07-18:
I can't reproduce this on my Mac but I'm on Apple Silicon (ARM). @austinharris is your Mac Intel/X86? I can believe there are errors there as we don't test on those.

chenbobby, 2023-08-21:
Hi, I am facing the same build error as @austinharris on macOS Apple Silicon (ARM). I have included notes on my error and details about my build environment below.

### Explanation of the problem

It seems that there are name conflicts between [mach/exception_types.h](https://opensource.apple.com/source/xnu/xnu-792/osfmk/mach/exception_types.h.auto.html) and [gdbremote/signals.hh](https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/ext/gdbremote/signals.hh#L171-L176) for the following names:

```
EXC_BAD_ACCESS
EXC_BAD_INSTRUCTION
EXC_ARITHMETIC
EXC_EMULATION
EXC_SOFTWARE
EXC_BREAKPOINT
```

[mach/exception_types.h](https://opensource.apple.com/source/xnu/xnu-792/osfmk/mach/exception_types.h.auto.html) uses these names in `#define` macros and sets them to integer constants (for example, `#define EXC_BAD_ACCESS 1`), but [gdbremote/signals.hh](https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/ext/gdbremote/signals.hh#L171-L176) uses these names as variants in the enum class `gem5::GDBSignal` (for example, `EXC_BAD_ACCESS = 145`). Due to the name conflict, [gdbremote/signals.hh](https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/ext/gdbremote/signals.hh#L171-L176) fails to build with the error `expected identifier`, because the macro expansion results in the code `1 = 145`, which is an invalid assignment.

### Caveats

@austinharris's build is using [mach/exception_types.h](https://opensource.apple.com/source/xnu/xnu-792/osfmk/mach/exception_types.h.auto.html) from libSystem, while my build is using [mach/exception_types.h](https://opensource.apple.com/source/xnu/xnu-792/osfmk/mach/exception_types.h.auto.html) from XCode (I have provided my error logs below). I'm _pretty_ sure the contents of these two files are the same, but not certain. If @BobbyRBruce is not facing the same errors on his Apple Silicon (ARM) machine, it may be because [mach/exception_types.h](https://opensource.apple.com/source/xnu/xnu-792/osfmk/mach/exception_types.h.auto.html) is somehow not being used in the build...?

### Possible solutions

1. Use an `#ifdef` guard so that we do not define `gem5::GDBSignal::EXC_*` for macOS. However, as @austinharris mentioned above, [mach/exception_types.h](https://opensource.apple.com/source/xnu/xnu-792/osfmk/mach/exception_types.h.auto.html) and [gdbremote/signals.hh](https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/ext/gdbremote/signals.hh#L171-L176) uses different values for these names. Moreover, [gdbremote/signals.hh](https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/ext/gdbremote/signals.hh#L171-L176) is using these names as variants within an enum class, not as a `#define` macro.
1. Change the names of the `gem5::GDBSignal::EXC_*` in [gdbremote/signals.hh](https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/ext/gdbremote/signals.hh#L171-L176) and across the gem5 codebase?
1. Somehow avoid compiling [mach/exception_types.h](https://opensource.apple.com/source/xnu/xnu-792/osfmk/mach/exception_types.h.auto.html) so that there are no macro naming conflicts?

Please let me know your thoughts or if I am missing anything. I'm happy to help open a PR to get this fixed and tested.

### Build Environment
CPU: Apple M1 Pro (2021)
OS: macos 13.3.1 (Ventura)
Compiler toolchain: gcc (Homebrew GCC 13.1.0) 13.1.0
Scons version: v4.5.2.120fd4f633e9ef3cafbc0fec35306d7555ffd1db
gem5 version: commit [f29bfc](https://github.com/gem5/gem5/tree/f29bfc0640c88a79eb7f94454ce31b3237ec0066)

Here's the build command that I ran:
```
$ python3 `which scons` build/X86/gem5.opt -j9
```
And here's the error log lines:
```
In file included from src/cpu/o3/probe/elastic_trace.cc:43:
In file included from src/cpu/o3/dyn_inst.hh:53:
In file included from src/cpu/checker/cpu.hh:51:
In file included from src/cpu/base.hh:60:
In file included from src/sim/system.hh:65:
In file included from src/sim/workload.hh:37:
ext/gdbremote/signals.hh:171:5: error: expected identifier
    EXC_BAD_ACCESS = 145, //could not access memory
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/mach/exception_types.h:68:33: note: expanded from macro 'EXC_BAD_ACCESS'
#define EXC_BAD_ACCESS          1       /* Could not access memory */
                                ^
In file included from src/cpu/o3/probe/elastic_trace.cc:43:
In file included from src/cpu/o3/dyn_inst.hh:53:
In file included from src/cpu/checker/cpu.hh:51:
In file included from src/cpu/base.hh:60:
In file included from src/sim/system.hh:65:
In file included from src/sim/workload.hh:37:
ext/gdbremote/signals.hh:172:5: error: expected identifier
    EXC_BAD_INSTRUCTION = 146, //illegal instruction/operand
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/mach/exception_types.h:72:33: note: expanded from macro 'EXC_BAD_INSTRUCTION'
#define EXC_BAD_INSTRUCTION     2       /* Instruction failed */
                                ^
In file included from src/cpu/o3/probe/elastic_trace.cc:43:
In file included from src/cpu/o3/dyn_inst.hh:53:
In file included from src/cpu/checker/cpu.hh:51:
In file included from src/cpu/base.hh:60:
[... 19 lines cut ...]
                                ^
In file included from src/cpu/o3/probe/elastic_trace.cc:43:
In file included from src/cpu/o3/dyn_inst.hh:53:
[... cut ...]
In file included from src/cpu/checker/cpu.hh:51:
In file included from src/cpu/base.hh:60:
In file included from src/sim/system.hh:65:
In file included from src/sim/workload.hh:37:
ext/gdbremote/signals.hh:176:5: error: expected identifier
    EXC_BREAKPOINT = 150, //breakpoint
    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/mach/exception_types.h:87:33: note: expanded from macro 'EXC_BREAKPOINT'
#define EXC_BREAKPOINT          6       /* Trace, breakpoint, etc. */
```

BobbyRBruce, 2023-08-21:
Yip! Can reproduce here! My issue was I was compiling with clang. Annoying Mac overrides 'gcc' and 'g++' with clang, so I needed to explicitly compile gem5 with `CXX=g++-13 scons build/X86/gem5.opt -j12` after `brew install gcc` for the error to work.

I'll work on fixing this.

(Short term solution for those needing to compile quick: Install and use clang. I think `CXX=clang++ scons build/X86/gem5.opt -j12` will work.

chenbobby, 2023-08-21:
It's okay! Thanks for taking a look at this. Hopefully we can resolve this soon.

Just want to confirm that using clang++ didn't fix the problem for me. 🥲 

For the record, my `clang++ --version` will output the following:
```
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: arm64-apple-darwin22.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

BobbyRBruce, 2023-08-22:
Hey @chenbobby (the other Bobby): Can you see if #209 fixes your issue?

I cannot recreate this on my end but I suspect your idea of renaming the `EXC_` would fix things. Can you see if incorporating this change fixes things for you?

I'm honestly not entirely sure how important these values in "ext/gdbremote/signals.hh" are. I hope this doesn't break anything.

powerjg, 2023-08-22:
Does anyone know why `mach/exception_types.h` is included in gem5's build? I can't imagine that gem5's build needs a MacOS-specific file.

chenbobby, 2023-08-22:
Hi @BobbyRBruce, #209 does indeed fix this issue for me. 🙌 🙏 

I'm now dealing with some build issues with protobuf 😅

BobbyRBruce, 2023-08-23:
@chenbobby : Great!

I've never tried using protobuf on a Mac before. If the issue persists then, if you have the time, please log it as an issue. If there are bugs or pitfalls here then I doubt you're the only one to encounter them.

FYI: protobuf isn't required for gem5 compilation but you do need it for trace generation and playback functionality. If you don't need that then it's safe to remove.

austinharris, 2023-08-24:
> Does anyone know why `mach/exception_types.h` is included in gem5's build? I can't imagine that gem5's build needs a MacOS-specific file.

I believe it is just included in every user-level program in order to handle exceptions such as system calls.

BobbyRBruce, 2023-08-24:
This issue is fixed by the merging of #209 (at least, for now, on the develop branch).

closed 2023-08-24 by BobbyRBruce