gem5/gem5 #97: printing bubble instruction in Minor CPU causes segfault

dbear496, 2023-07-17:
It appears that printing a bubble instruction in the Minor CPU causes a segfault. 

Steps to reproduce:
1. Run gem5 with the Minor CPU and the 'Branch' or 'Fetch' debug flags.
2. Cause an interrupt so that [minor::Execute::takeInterrupt](https://github.com/gem5/gem5/blob/af72b9ba580546ac12ce05bfaac3fd53fa8699f4/src/cpu/minor/execute.cc#L422) is called.
3. Observe a segfault at [src/cpu/minor/dyn_inst.cc:116](https://github.com/gem5/gem5/blob/af72b9ba580546ac12ce05bfaac3fd53fa8699f4/src/cpu/minor/dyn_inst.cc#L116)

If the 'Branch' debug flag is used, then the segfault is caused by the DPRINTF statement at [src/cpu/minor/execute.cc:320](https://github.com/gem5/gem5/blob/af72b9ba580546ac12ce05bfaac3fd53fa8699f4/src/cpu/minor/execute.cc#L320); otherwise, if the 'Fetch' debug flag is used, then the segfault is caused by the DPRINTF statement at [src/cpu/minor/fetch1.cc:513](https://github.com/gem5/gem5/blob/af72b9ba580546ac12ce05bfaac3fd53fa8699f4/src/cpu/minor/fetch1.cc#L513).

I am using gem5 version 22.1.0.0, so I apologize if this is already fixed in the current version.

BobbyRBruce, 2023-07-19:
I wrote the following gem5 Python configuration script to try to reproduce this:

```py
from gem5.utils.requires import requires
from gem5.components.boards.x86_board import X86Board
from gem5.components.memory.single_channel import SingleChannelDDR3_1600
from gem5.components.processors.simple_processor import SimpleProcessor
from gem5.components.processors.cpu_types import CPUTypes
from gem5.isas import ISA
from gem5.coherence_protocol import CoherenceProtocol
from gem5.simulate.simulator import Simulator
from gem5.components.cachehierarchies.classic.no_cache import NoCache
from gem5.resources.workload import Workload

requires(isa_required=ISA.X86)

cache_hierarchy = NoCache()

memory = SingleChannelDDR3_1600(size="3GB")

processor = SimpleProcessor(cpu_type=CPUTypes.MINOR, isa=ISA.X86, num_cores=1)

board = X86Board(
    clk_freq="3GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=cache_hierarchy,
)

workload = Workload("x86-ubuntu-18.04-boot")
board.set_workload(workload)

simulator = Simulator(board=board)
simulator.run()
```

On v23.0.0.1 I ran:

```sh
rm -rf build
scons build/ALL/gem5.opt -j`nproc`
./build/ALL/gem5.opt --debug-flags=Fetch,Branch <config scriot>
```

I let it run for 15 minutes and I never encountered this bug.

I think ran it on v22.1.0.0, again, for about 15 minutes, and nothing was encountered.

Do you have a way to reproduce this? If so, can you see if it still occurs in v23.0. And if so, can you share it here?

dbear496, 2023-07-19:
Using v23.0.0.1 and the configuration script you shared, it does not appear that an interrupt is ever invoked, and hence, `takeInterrupt` is never called. However, I was able to reproduce the issue by faking an interrupt with gdb with the following steps:
1. Set a breakpoint at `X86ISA::Interrupts::checkInterrupts`, and continue.
2. When the breakpoint is hit, run the gdb command: `print pendingUnmaskableInt = pendingNmi = true`.
3. Disable the breakpoint, and continue.

BobbyRBruce, 2023-07-19:
Thanks for providing that. I'll try to find some time to look into this. I'm just wondering, is there a workload you have, a binary or something, that will invoke this? The gold-standard would be evidence this state is reachable in a simulation run.

dbear496, 2023-07-19:
I came across the issue while implementing the NIOS II ISA in gem5, so I am providing my own `NiosiiISA::Interrupts::checkInterrupts` method. And to be really honest with you, I do not know how interrupts would normally be invoked in gem5 with other ISAs. My current implementation hard-wires a custom UART module into the ISA, and I noticed that I get a segfault when `NiosiiISA::Interrupts::checkInterrupts` returns true. This is a temporary solution until I implement the UART as a PioDevice. However, at the end of the day, `NiosiiISA::Interrupts::checkInterrupts` must at some point return true to invoke the UART driver interrupt handler even if I properly use the PioDevice framework.

So to be clear, I do not think this issue is specific to the (janky) system I have running. I believe any interrupt whatsoever would cause the segfault. So the "gold-standard" would be whatever normal conditions would cause `takeInterrupt` to be called -- perhaps receiving input from a device.

As a workaround to get my simulation to run, I modified `operator <<(std::ostream &os, const MinorDynInst &inst)` to check that `inst.pc` is non-null before dereferencing it.

powerjg, 2023-07-20:
Ah, I see the problem. It looks like https://gem5-review.googlesource.com/c/public/gem5/+/52047 caused this bug.

I think we should make sure that `pc` is not null whenever it is dereferenced.

powerjg, 2023-07-20:
@dbear496, it would be great if you could test #107 to see if it fixes the problem for you given that we couldn't easily recreate the segfault.

dbear496, 2023-07-21:
> @dbear496, it would be great if you could test #107 to see if it fixes the problem for you given that we couldn't easily recreate the segfault.

Yes, that fixes the issue.

PR #107 (2023-07-26): cpu-minor: Check pc valid before printing
In https://gem5-review.googlesource.com/c/public/gem5/+/52047 inst.pc was changed from an object to a pointer. It is possible that this pointer is null (e.g., if there is an interrupt and there is a bubble). Make sure to check that it's not null before printing.

I believe that other places this pointer is dereferenced without an explicit null check are safe, but I'm not certain.

Should fix #97 

Change-Id: Idbe246cfdb62d4d75416d41b451fb3c076233bbc
commit 7601fcfb (2023-07-26): cpu-minor: Check pc valid before printing (#107)

In https://gem5-review.googlesource.com/c/public/gem5/+/52047 inst.pc
was changed from an object to a pointer. It is possible that this
pointer is null (e.g., if there is an interrupt and there is a bubble).
Make sure to check that it's not null before printing.

I believe that other places this pointer is dereferenced without an
explicit null check are safe, but I'm not certain.

Should fix #97 

Change-Id: Idbe246cfdb62d4d75416d41b451fb3c076233bbc
closed 2023-07-26 by BobbyRBruce