gem5/gem5 #100: `BaseCache::recvTimingResp` can trigger in `getTarget()` assertion error due to MSHR in `senderState` having no targets

BobbyRBruce, 2023-07-19:
This was identified by Eliot moss on our gem5-users mailing list (https://www.mail-archive.com/gem5-users@gem5.org/msg21771.html). Copy and pasted verbatim:

```txt
Dear gem5-ers -

I've run across something, which unfortunately I cannot reliably repeat, that
suggests an oversight in the code of src/mem/cache/base.cc.  In particular, it
appears that a HardPFResp can arrive where the MSHR remembered in the packet's
senderState no longer has any targets.  This causes this line to fail with an
assertion error in getTarget():

const QueueEntry::Target *initial_tgt = mshr->getTarget();

Presumably some suitable conditionalization on mshr->hasTargets() could be
used to fix things, unless the problem is that somehow the target(s)
disappeared when they should not have.  My suspicion is that something to do
with snooping by another cache caused the target to go away, and the situation
should just be ignored, but I did not want to attempt a fix along the lines of
testing hasTargets() without further confirmation.  There is also the question
of what to do about the stats in this case, there being no obvious basis for
determining a latency.  [We could change the senderState to include the time
the prefetch began, though, and use HardPFReq as the command.]

Regards - Eliot Moss
```

The `` line mentioned in the email referred to:
https://github.com/gem5/gem5/blob/af72b9ba580546ac12ce05bfaac3fd53fa8699f4/src/mem/cache/base.cc#L528

eliotmoss, 2023-07-19:
The configuration where this seems to have happened (it's not an easily repeated thing) is ARM DerivO3 with two processors.  The setup is not entirely symmetrical - one "host" processor runs Linux has has at least an L2 cache.  The other "embedded" processor is more "bare metal" and has at least an L1 cache.  Their point of mutual connection is the system's membus.

If desired I can propose a patch that avoids doing getTarget when there are no targets, and supplies something reasonable for incrementing the statistics (namely the command of the packet received).

BobbyRBruce, 2023-07-19:
My worry is this could symptomatic of a bigger bug. If we work around the assertion we may struggle to ever detect it. I'm tempted to wait until we can reproduce it, explain why it happens, or someone has a good explanation as to why it's ok for the MSHR in `senderState` to have no targets at this point.

eliotmoss, 2023-07-19:
That's reasonable, Bobby.  While I've dug into - and extended - various parts of cache behavior, prefetching is one part I have not.  I'll see if I can make some time to dig out how targets might disappear while a prefetch is pending.  My guess is that some kind of snoop may have caused contents, including some outstanding MSHRs / targets, to be dropped, but that just a starting point for exploration.  Another angle would be an eviction.  Happy if anybody else chooses to look into it as well, of course!

eliotmoss, 2023-07-19:
I wonder about this line:

https://github.com/gem5/gem5/blob/af72b9ba580546ac12ce05bfaac3fd53fa8699f4/src/mem/cache/base.cc#L320

closed 2024-01-08 by ivanaamit