gem5/gem5 #112: Build error when linking gem5.opt

GeneralKinseng, 2023-07-23:
Hi,

I encountered an error when building gem5, saying "error adding symbols: DSO missing from command line".
How to solve this?

>  [    LINK]  -> RISCV/gem5.opt
> /home/kinseng/chipyard/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.2.0/../../../../x86_64-conda-linux-gnu/bin/ld: build/RISCV/proto/inst_dep_record.pb.o: undefined reference to symbol '_ZN4absl12lts_2023012512log_internal17MakeCheckOpStringIllEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEET_T0_PKc'
> /home/kinseng/chipyard/.conda-env/bin/../lib/gcc/x86_64-conda-linux-gnu/12.2.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/kinseng/chipyard/.conda-env/lib/libabsl_log_internal_check_op.so.2301.0.0: error adding symbols: DSO missing from command line
> collect2: error: ld returned 1 exit status
> scons: *** [build/RISCV/gem5.opt] Error 1
> scons: building terminated because of errors.

powerjg, 2023-07-24:
It looks like you don't have the protobuf development environment installed correctly. On Ubuntu you can use `apt install libprotobuf-dev protobuf-compiler libprotoc-dev libgoogle-perftools-dev`.

I have personally never used a conda environment to build gem5. I would suggest using a docker container if you still have problems. 

See https://www.gem5.org/documentation/general_docs/building#docker for more information.

BobbyRBruce, 2023-07-26:
@GeneralKinseng : Did this fix the issue for you?

GeneralKinseng, 2023-07-26:
The build seems to have conflicts with conda environment. I have disabled the environment and rebuild gem5. It works perfectly. Thank you for your help.

closed 2023-07-26 by GeneralKinseng

scratch-er, 2023-08-01:
I encountered a similar error on arch linux. However, I'm not using an conda environment and I have protobuf installed.

Here is the full build log

```
scons -j12 build/RISCV/gem5.opt
scons: Reading SConscript files ...

You're missing the pre-commit/commit-msg hooks. These hook help to ensure your
code follows gem5's style rules on git commit and your commit messages follow
our commit message requirements. This script will now install these hooks in
your .git/hooks/ directory.
Press enter to continue, or ctrl-c to abort:
 Cannot find 'pre-commit'. Please ensure all Python requirements are 
installed. This can be done via 'pip install -r requirements.txt'.
It is strongly recommended you install the pre-commit hooks before working with gem5. Do you want to continue compilation (y/n)?
y
Mkdir("/home/scratcher/Workspace/wasm-task3.5/gem5/build/RISCV/gem5.build")
Checking for linker -Wl,--as-needed support... (cached) yes
Checking for compiler -gz support... (cached) yes
Checking for linker -gz support... (cached) yes
Info: Using Python config: python3-config
Checking for C header file Python.h... (cached) yes
Checking Python version... (cached) 3.11.3
Checking for accept(0,0,0) in C++ library None... (cached) yes
Checking for zlibVersion() in C++ library z... (cached) yes
Checking for C library tcmalloc_minimal... (cached) yes
Building in /home/scratcher/Workspace/wasm-task3.5/gem5/build/RISCV
Using saved variables file(s) /home/scratcher/Workspace/wasm-task3.5/gem5/build/RISCV/gem5.build/variables
Checking size of struct kvm_xsave ... (cached) yes
Checking for C header file fenv.h... (cached) yes
Checking for C header file png.h... (cached) yes
Checking for clock_nanosleep(0,0,NULL,NULL) in C library None... (cached) yes
Checking for C header file valgrind/valgrind.h... (cached) no
Checking for pkg-config package hdf5-serial... (cached) no
Checking for pkg-config package hdf5... (cached) yes
Checking for H5Fcreate("", 0, 0, 0) in C library hdf5... (cached) yes
Checking for H5::H5File("", 0) in C++ library hdf5_cpp... (cached) yes
Checking for C header file linux/kvm.h... (cached) yes
Checking for timer_create(CLOCK_MONOTONIC, NULL, NULL) in C library None... (cached) yes
Checking for member exclude_host in struct perf_event_attr...(cached) yes
Checking for C header file linux/if_tun.h... (cached) yes
Checking for shm_open("/test", 0, 0) in C library None... (cached) yes
Checking for pkg-config package protobuf... (cached) yes
Checking for GOOGLE_PROTOBUF_VERIFY_VERSION in C++ library protobuf... (cached) yes
Checking for backtrace_symbols_fd((void *)1, 0, 0) in C library None... (cached) yes
Checking whether __i386__ is declared... (cached) no
Checking whether __x86_64__ is declared... (cached) yes
Checking for compiler -Wno-self-assign-overloaded support... (cached) yes
Checking for linker -Wno-free-nonheap-object support... (cached) yes
scons: done reading SConscript files.
scons: Building targets ...
 [VER TAGS]  -> RISCV/sim/tags.cc
 [    LINK]  -> RISCV/gem5.opt
/usr/bin/ld: build/RISCV/cpu/o3/probe/elastic_trace.o: undefined reference to symbol '_ZN4absl12lts_2023012512log_internal17MakeCheckOpStringIllEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEET_T0_PKc'
/usr/bin/ld: /usr/lib/libabsl_log_internal_check_op.so.2301.0.0: error adding symbols: DSO missing from command line
collect2: error：ld returned 1
scons: *** [build/RISCV/gem5.opt] Error 1
scons: building terminated because of errors.

```

OdnetninI, 2023-11-08:
Just for helping anyone with this issue (like @scratch-er ):

Downgrading protobuf and python-protobuf to version 23.4 on archlinux fixes the issue.
Here is the command for downgrading them using the ArchLinux Archive Repository

```
sudo pacman -U https://archive.archlinux.org/packages/p/protobuf/protobuf-23.4-1-x86_64.pkg.tar.zst https://archive.archlinux.org/packages/p/python-protobuf/python-protobuf-23.4-1-x86_64.pkg.tar.zst
```

Remember that every update will revert it.

Also, take into account that other software that depends on protobuf could be affected.

codetang-2417, 2023-12-16:
> 只是为了帮助任何人解决这个问题（例如@scratch-er）：
> 
> 在 archlinux 上将 protobuf 和 python-protobuf 降级到版本 23.4 可以修复该问题。 这是使用 ArchLinux 存档存储库降级它们的命令
> 
> ```
> sudo pacman -U https://archive.archlinux.org/packages/p/protobuf/protobuf-23.4-1-x86_64.pkg.tar.zst https://archive.archlinux.org/packages/p/python-protobuf/python-protobuf-23.4-1-x86_64.pkg.tar.zst
> ```
> 
> 请记住，每次更新都会恢复它。
> 
> 另外，请考虑依赖于 protobuf 的其他软件可能会受到影响。

我在manjaro上尝试了这个方法，但是并不能解决问题。但我发现，从最新protobuf降级到23.4，protobuf的依赖库abseil并不会自动降级到对应的版本，这会导致protobuf complier(protoc)不能正常工作，这样的话gem5编译时会直接跳过tracing support，然后编译出一个可用的版本，只是不支持tracing support。

js4ngu, 2024-08-30:
A similar error occurs to me.
I don't know how to do it.
Can I get some help?

```
 [VER TAGS]  -> RISCV/sim/tags.cc
 [    LINK]  -> RISCV/gem5.opt
/usr/bin/ld: build/RISCV/proto/inst_dep_record.pb.o: in function `google::protobuf::MessageLite::~MessageLite()':
/home/jongsang/src/protobuf/include/google/protobuf/message_lite.h:171: undefined reference to `google::protobuf::internal::InternalMetadata::~InternalMetadata()'
/usr/bin/ld: /home/jongsang/src/protobuf/include/google/protobuf/message_lite.h:171: undefined reference to `google::protobuf::internal::InternalMetadata::~InternalMetadata()'
/usr/bin/ld: /home/jongsang/src/protobuf/include/google/protobuf/message_lite.h:171: undefined reference to `google::protobuf::internal::InternalMetadata::~InternalMetadata()'
/usr/bin/ld: /home/jongsang/src/protobuf/include/google/protobuf/message_lite.h:171: undefined reference to `google::protobuf::internal::InternalMetadata::~InternalMetadata()'
/usr/bin/ld: /home/jongsang/src/protobuf/include/google/protobuf/message_lite.h:171: undefined reference to `google::protobuf::internal::InternalMetadata::~InternalMetadata()'
/usr/bin/ld: build/RISCV/proto/inst_dep_record.pb.o:/home/jongsang/src/protobuf/include/google/protobuf/message_lite.h:171: more undefined references to `google::protobuf::internal::InternalMetadata::~InternalMetadata()' follow
collect2: error: ld returned 1 exit status
scons: *** [build/RISCV/gem5.opt] Error 1
scons: building terminated because of errors.

```