gem5/gem5 #123: Reporting traps (faults) to GDB in SE mode is broken

janvrany, 2023-07-26:
While using gem5 in SE mode I realized that reporting traps to GDB (if connected) is broken / missing at least on 
POWER / RISC-V. 

Consider POWER: https://github.com/gem5/gem5/blob/develop/src/arch/power/faults.cc#L42-L63. `trapToGdb` returns `true` is GDB is connected *and* trap has been reported, otherwise `false`. But the code in `TrapFault::invoke` panics *if* there's a GDB connected and does nothing if there's no GDB at all! I'm pretty sure the logic should be inverse. 

On RISC-V (and possibly other architectures as well) there's no such logic at all, look  at https://github.com/gem5/gem5/blob/develop/src/arch/riscv/faults.cc, so on RISC-V one does not see software breakpoints hitting at all when connecting GDB to debug program running in gem5.

powerjg, 2023-07-26:
Seems like a good bug to fix! Thanks for volunteering!

shingarov, 2023-07-26:
I just stepped through it in the debugger — you are absolutely right @janvrany !  Looks like this is a recent regression introduced in https://gem5-review.googlesource.com/c/public/gem5/+/63534

janvrany, 2023-07-27:
@powerjg Thanks, it's "just" a matter of me finding a scrap of time. I'm currently swamped by other work : - ( Hopefully in a week or two,..

BobbyRBruce, 2023-08-22:
Fixed by #166

closed 2023-08-22 by BobbyRBruce