gem5/gem5 #131: GPU Failure in the Daily Tests

mkjost0, 2023-07-27:
Currently, the long gpu tests are failing in our dailies.  This failure can be seen here (https://github.com/gem5/gem5/actions/runs/5653802586/job/15319989256). In order to recreate this, you can run the following code below, though all 4 suites in the tests fail.  These fail after the inclusion of pull request 101 (https://github.com/gem5/gem5/pull/101)

```
scons build/VEGA_X86/gem5.opt
build/VEGA_X86/gem5.opt configs/example/ruby_gpu_random_test.py --test-length 50000 --num-dmas 0
```

When you run this, we receive the following output, followed by a backtrace output:

```
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (1024 Mbytes)
src/cpu/testers/gpu_ruby_test/protocol_tester.cc:140: warn: Data race check is not enabled
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
      1: global: Episode 0
src/cpu/testers/gpu_ruby_test/address_manager.cc:391: warn: Expected return values are:
         0
src/cpu/testers/gpu_ruby_test/tester_thread.cc:388: warn: GpuWavefront(TesterThread ID = 0, CU ID = 0): Atomic Op returned unexpected value
        Episode 0
        Lane ID 0
        Address [0x31884, line 0x31880]
        Atomic Op's return value 1

src/cpu/testers/gpu_ruby_test/protocol_tester.cc:345: panic: GPU Ruby Tester: Failed!
Memory Usage: 1358852 KBytes
Program aborted at tick 1537
```

abmerop, 2023-07-27:
Seem to be a few things going on here. This test changed entirely because now all atomics are being done at TCC instead of SD like there were before the PR.

1) wdb_writeDirtyBytes is reseting cache line value to 0 every time.  The data value in an atomic request is not valid I believe. This should be replaced with a new action which only OR's the writeMask.  Setting the mask is needed as it is used by atomicPartial.
2) The value returned is the value *after* the atomic. GPUs return the value *before* the atomic. This can be fixed easily by swapping ar_sendAtomicResponse and pa_performAtomic.
3) There is some other bug after making those changes related to AtomicDone in M state transition missing.

Another aside: This test is using a random time-based seed and the tester does not print out what the value is. This could make it very difficult in the future to reproduce the test.  It happens to be fine here as every seed will fail for sure.  The protocol tester should print out what the seed is or the github action command should explicitly add --random-seed=<nonzero value>.

abmerop, 2023-07-27:
We should probably revert to the old behavior to fix the test ASAP.  I have a pull here: https://github.com/gem5/gem5/pull/133

The 3rd bug I mentioned above is actually because there is no support to evict a cache line for an atomic needs when the cache set is full.  I suspect it will take some days to implement that.

closed 2023-07-27 by powerjg

mattsinc, 2023-07-27:
@DanKouch are #2 and #3 addressed by commits that are pending in your chain?

BobbyRBruce, 2023-07-27:
> Another aside: This test is using a random time-based seed and the tester does not print out what the value is. This could make it very difficult in the future to reproduce the test. It happens to [...]

Could we remove this non-determinisim from the test? I don't mind adding a `--random-seed=` flag but I don't think any RNG in gem5 should be seeded on time/date or anything that isn't fixed (or determined via something explicitly passed as a parameter).

powerjg, 2023-07-27:
@BobbyRBruce, let's move this discussion to #138.

commit ff7e67ee (2023-07-27): cpu: Set SLC bit for GPU tester

This fixes #131 by reverting to the old behavior of performing all
atomics at the system level. To do this the SLC bit needs to be set for
all atomic requests.

Change-Id: I63f4e449be1b02c933832d09700237f8c8026f4c