gem5/gem5 #138: Daily GPU tests fail to output the random seed used

mkjost0, 2023-07-27:
As mentioned in pull request #131 , the workflow for dailies needs to be altered so that the random seed used for tests is output in order to allow for us to recreate tests.  More specifics copied from the PR below:

"This test is using a random time-based seed and the tester does not print out what the value is. This could make it very difficult in the future to reproduce the test. It happens to be fine here as every seed will fail for sure. The protocol tester should print out what the seed is or the github action command should explicitly add --random-seed=."

BobbyRBruce, 2023-07-27:
Why do we have non-determinism anywhere in gem5? To me this isn't a problem with the tests, there shouldn't be anything truely random going on. The "time-based seed" should be replaced with something static.

mkjost0, 2023-07-27:
Yeah, I was thinking of just altering the files in the tests directory to use `--random-seed=` so that we don't have to make more changes to the workflow directory.  If that's okay, I can just assign this to myself and get that done.

mattsinc, 2023-07-27:
I had asked @DanKouch to do the same @mkjost0 , but if you want to get to it today, please go ahead.

@abmerop (or really, Brad) will have to comment more on @BobbyRBruce 's original question though, as I was not involved with that work.

powerjg, 2023-07-27:
Changing the command line parameter is the best thing to do in the short term (I think). 

Longer term, the problem is this line: https://github.com/gem5/gem5/blob/af72b9ba580546ac12ce05bfaac3fd53fa8699f4/src/cpu/testers/gpu_ruby_test/protocol_tester.cc#L145

We *should* be using `mt_random` here instead of `srand`. We should always use gem5's random number generator so that we have determinism by default.

Then, when we want to have multiple different seeds used, we can use (from *python*) `m5._m5.core.seedRandom(random.randint(0,100000))` (or something like this). We should expose this `seedRandom` function better, though.

Anyway, the "right" solution is a bit deeper than is worth it at this point. For now, let's just set the command line parameter.

BobbyRBruce, 2023-07-28:
> Changing the command line parameter is the best thing to do in the short term (I think).
>
> Longer term, the problem is this line: [...]

I had a go at this in #140. I'm not an expert on this part of the codebase, but I think my implementation of `random_mt` works.

powerjg, 2023-07-31:
@mkjost0, can you check the status of this since #140 has been merged?