gem5/gem5 #144: RVV vset*vl* instructions execute bad in out-of-order (O3) mode

vmskv, 2023-07-31:
**Describe the bug**
RVV vset\*vl\* instructions can be executed bad in out-of-order (O3) mode. If vset\*vl\* is in mis-predicted branch it changes the 'VL' value in decoder and the vector instructions afterwards are decoded/executed in correspondence with 'VL' value which was set in squashed instruction.

**Affects version**
Found in [arch-riscv: Relation chain on RVV support #83](https://github.com/gem5/gem5/pull/83)  PR.

**To Reproduce**
The reproducer is in https://github.com/vmskv/gem5/tree/vmskv/rvv_bad_o3_vsetvl_execuion/riscv_o3_bad_vsetvl_execution_reproducer 
The steps to reproduce are in [README.txt](https://github.com/vmskv/gem5/blob/vmskv/rvv_bad_o3_vsetvl_execuion/riscv_o3_bad_vsetvl_execution_reproducer/README.txt)

**Expected behavior**
If vset\*vl\* is squashed the last previous correct 'VL' value should be used to decode the vector instructions.

**Host Operating System**
Ubuntu 20.04

**Guest ISA**
RISC-V.

powerjg, 2023-07-31:
cc: @hnpl @adriaarmejach

hnpl, 2023-08-02:
I have an implementation that modifies the PCState for vl and vtype speculation, which is relatively straightforward. If I remember correctly, this version works with O3CPU but it does not work with MinorCPU.

https://github.com/hnpl/plct-gem5-1/commits/rvv-cpu

rogerchang23424, 2023-08-25:
#228

BobbyRBruce, 2023-09-14:
Fixed by #228.

closed 2023-09-14 by BobbyRBruce

mshiryaev, 2023-09-15:
@BobbyRBruce - could you please clarify when RVV support for O3 CPU (including changes in https://github.com/gem5/gem5/pull/228.) will be available in gem5 release? As far as I see the latest one (v23.0.1.0) doesn't support O3/RVV. Any rough estimation would be helpful.

BobbyRBruce, 2023-09-15:
> @BobbyRBruce - could you please clarify when RVV support for O3 CPU (including changes in #228.) will be available in gem5 release? As far as I see the latest one (v23.0.1.0) doesn't support O3/RVV. Any rough estimation would be helpful.

@rogerchang23424 : Am i correct on thinking that on the develop branch the O3 CPU supports RVV? I assume since #228 is merged this is the case?

@mshiryaev: The next gem5 release (v23.1) should be in the next couple of months and certainly by the end of the year. If you wish to use gem5 in it's current development state you can do so by checking out that branch:

```sh
git clone https://github.com/gem/gem5.git
cd gem5
git switch develop
```

Generally `develop` is in a good state and usable but it just comes with no guarantees everything is working and fully implemented.

rogerchang23424, 2023-09-16:
> @rogerchang23424 : Am i correct on thinking that on the develop branch the O3 CPU supports RVV? I assume since https://github.com/gem5/gem5/pull/228 is merged this is the case?

Yes, I have tests the vector instructions with examples of [rvv-intrinsic-doc](https://github.com/riscv-non-isa/rvv-intrinsic-doc) , [riscv_o3_bad_vsetvl_execution_reproducer](https://github.com/vmskv/gem5/tree/vmskv/rvv_bad_o3_vsetvl_execuion/riscv_o3_bad_vsetvl_execution_reproducer) ,and https://github.com/riscv-software-src/riscv-tests/pull/400 and work well in O3 CPU. We can assume the issue is solved by #228, but  I think the issue may still triggered in other condition.

mshiryaev, 2023-09-16:
@BobbyRBruce , @rogerchang23424 - thanks for clarification!