gem5/gem5 #146: RISC-V fails to compile

janvrany, 2023-07-31:
I'm trying to compile current `develop` (618b2a60d) for RISC-V (`scons build/RISCV/gem5.debug`) and the compilation fails:

```
 [     CXX] RISCV/debug/PMP.cc -> .do
 [ISA DESC] src/arch/riscv/isa/main.isa -> RISCV/arch/riscv/generated/decoder-g.cc.inc, RISCV/arch/riscv/generated/decoder-ns.cc.inc, RISCV/arch/riscv/generated/decode-method.cc.inc, RISCV/arch/riscv/generated/decoder.hh, RISCV/arch/riscv/generated/decoder-g.hh.inc, RISCV/arch/riscv/generated/decoder-ns.hh.inc, RISCV/arch/riscv/generated/exec-g.cc.inc, RISCV/arch/riscv/generated/exec-ns.cc.inc, RISCV/arch/riscv/generated/decoder.cc, RISCV/arch/riscv/generated/inst-constrs.cc, RISCV/arch/riscv/generated/generic_cpu_exec.cc
In file included from /redacted/gem5/build/RISCV/arch/riscv/isa/main.isa:1:
In file included from /redacted/gem5/build/RISCV/arch/riscv/isa/formats/formats.isa:1:
At 0:
Error including file "/redacted/gem5/build/RISCV/arch/riscv/isa/formats/bs.isa"
scons: *** [build/RISCV/arch/riscv/generated/decoder.cc] Explicit exit, status 1
scons: building terminated because of errors.
*** Summary of Warnings ***
Warning: Couldn't find HDF5 C++ libraries. Disabling HDF5 support.
```

Reverting commit [4dccd7dd6 - arch-riscv: Add BS format isa](4dccd7dd6) fixes the problem for me.

janvrany, 2023-07-31:
FYI: @rogerchang23424

mkjost0, 2023-07-31:
Hi, we've tried recreating this on both AMD and Apple Arm systems and they've compiled fine.  Can you try deleting your build directory and recompiling to double-check that this is still an issue for you?

rogerchang23424, 2023-08-01:
Hi @janvrany,

It seems the repository is missing the bs.isa file in the `src/arch/riscv/isa/formats` folder. Cloud you check the folder again?

Thanks

closed 2023-08-01 by janvrany