gem5/gem5 #148: Enable loading of local gem5 Resources data by a simple passing of the data's JSON file path

BobbyRBruce, 2023-08-01:
As of v23.0  the user can specify custom gem5 Resources data locations (i.e., JSON files or database which store the gem5-resources meta-data: IDs, source information, download links, etc.) by creating a configuration file named "gem5-config.json" in the CWD or by using the `GEM5_CONFIG` environment variable to specify the location of the config. If neither a "gem5-config.json" file is present in the CWD or the `GEM5_CONFIG` environment variable is unset, the "[src/python/gem5_default_config.py](https://github.com/gem5/gem5/blob/48b4788bbe7a091b917aed51236bca2b0456cf37/src/python/gem5_default_config.py)" defaults are used (our MongoDB instance).

The reason a configuration file is needed is because gem5, as of v23.0, supports resource data being stored in a JSON file or in a MongoDB database. v23.0 also allows for multiple data sources to be merged, so users may, for example, use the gem5 project's MongoDB database as well as a local JSON file. In this case the configuration would look something like:

```json
{
    "sources": {
        "gem5-resources-local-json": {
            "url": "<path to resources.json>",
            "isMongo": false
        },
        "gem5-resources": {
            "dataSource": "gem5-vision",
            "database": "gem5-vision",
            "collection": "resources",
            "url": "https://data.mongodb-api.com/app/data-ejhjf/endpoint/data/v1",
            "authUrl": "https://realm.mongodb.com/api/client/v2.0/app/data-ejhjf/auth/providers/api-key/login",
            "apiKey": "OIi5bAP7xxIGK782t8ZoiD2BkBGEzMdX3upChf9zdCxHSnMoiTnjI22Yw5kOSgy9",
            "isMongo": True,
        }
    }
}
```

This is an ok solution for the criteria of handling two different resources data source types and specifying multiple locations to be used in unison . However, the downside of this new approach is it requires creation of the configuration file every time the user wishes to use resources other than those specified in the project's MongoDB instance. I've found that 9 times out of 10 if I have a local gem5 Resources data i want to be loaded into gem5 then it's in the JSON format and I'm not interested in loading other sources. Prior to gem5 v23.0 users could specify a local resource JSON file to load using the `GEM5_RESOURCE_JSON` environment variable, but in v23.0 this was removed in favor of the configuration file.

I think given passing a single resources JSON file is so common, there should be an simple way to do it which doesn't require creation of a configuration file. The user should only need to pass the path to the JSON file.

I can think of two solutions for this problem:

1. Re-implement the `GEM5_RESOURCE_JSON` environment variable and, if set, the gem5 standard library uses it's value as the path to load the resources JSON from alone, ignoring all other data sources. (Perhaps there should also be another environment variable (`GEM5_RESOURCE_JSON_APPEND`?) which appends the resources JSON file to the default MongoDB database. This could also be useful).
2. Add a parameter to the [`obtain_resources`](https://github.com/gem5/gem5/blob/48b4788bbe7a091b917aed51236bca2b0456cf37/src/python/gem5/resources/resource.py#L557) function which enables the passing of a file path to a resources JSON file. If set, this local JSON file would be used (note: work would need done to ensure this doesn't interference with the `clients` parameter in this function).

Harshil2107, 2023-08-01:
Current possible solution:
Allow users to set the following environment variables:
- ```GEM5_RESOURCE_JSON```
- ```GEM5_RESOURCE_JSON_APPEND```

if  ```GEM5_RESOURCE_JSON``` is set:
- If the above flag is set then all configs are ignored and just the file in the environment variable is used.
- A warning or error message will be sent noting that configs are ignored and the JSON file is used.

if ```GEM5_RESOURCE_JSON_APPEND``` is set:
- The JSON file is merged into the selected config.
- An inform statement stating which config is used and which file is merged will be thrown.

If both of the above mentions environment variables are set then an error is thrown.

**Note: None of the config files are actually changed.**

BobbyRBruce, 2023-08-01:
> Current possible solution: Allow users to set the following environment variables:
>
> * `GEM5_RESOURCE_JSON`
> * `GEM5_RESOURCE_JSON_APPEND`
>
> if `GEM5_RESOURCE_JSON` is set:
>
> * If the above flag is set [...]

I think my vote here is to throw a warning when using the `GEM5_RESOURCE_JSON` saying all other configs are ignored. A fatal/error is too harsh as the only way to resolve it would be for the user to remove their config, which doesn't seem like good design. The info output for the `GEM5_RESOURCE_JSON_APPEND` should be thorough as to what's happening.

closed 2023-08-02 by BobbyRBruce