gem5/gem5 #159: Is there a limit to the number of in-flight memory instructions in Minor CPU?

dbear496, 2023-08-04:
I have configured the Minor CPU with only two FUs (ALU and memory). Because of this, the size of the `inFlightInsts` queue is very small, and I am getting a warning: `build/NIOSII/cpu/minor/buffers.hh:442: warn: system.cpu.execute.inFlightInsts0: No space to push data into queue of capacity 2, pushing anyway`.

Upon further inspection, it seems that this is caused by 1) construction of the `inFlightInsts` queue [assumes](https://github.com/gem5/gem5/blob/48b4788bbe7a091b917aed51236bca2b0456cf37/src/cpu/minor/execute.cc#L141) that each FU will have a maximum of `opLat` instructions in flight and 2) memory instructions are considered "in flight" after exiting the FU until receiving a memory response. `MinorDefaultMemFU` has a configured `opLat` of 1, but I can see via the MinorTrace debug flag that more than 1 memory instruction may be in the `inFlightInsts` queue at a time.

So I have a question: Is there some mechanism to limit the number of memory instructions in the `inFlightInsts` queue? Since I have done some tinkering, I want to make sure I have not broken such a mechanism if one exists. In my inspection of Execute::issue, Execute::commit, and Execute::commitInst, I did not see such a limit being enforced. If the above warning is supposed to be "normal", then perhaps it should be a warn_once so it does not spam the terminal.

seanzw, 2023-08-04:
After some investigation I think your understanding is correct: the `inFlightInsts` queue is sized according to the number of FUs and their latency, while load instructions are considered "in flight" until they received the response. This mismatch causes this warning. And I did not find the place where it would limit the number of inflight instructions directly. However, `executeLSQTransfersQueueSize` should do the job for you -- if the LSQ has no slot in the transfer queue, the memory instruction can not be issued, see "LSQ::canSendToMemorySystem()" and related code.

So back to your question: can you try play with these parameters and see if the number of in flight memory request is limited? As for `inFlightInsts`, I think it would be nice to fix it (but should not affect the simulation correctness), or document this behavior better.

dbear496, 2023-08-07:
Thanks. I'll see what I can do with that. I agree it would be nice to fix this behavior -- at least to increase the size of the inFlightInsts queue so the warning is less likely to occur.

closed 2023-08-07 by dbear496