gem5/gem5 #161: Adding VLEN and ELEN dynamic configuration to RISCV V 1.0 implementation

alvaromorenomv, 2023-08-07:
Hi!
I want to ask if someone is working on the dynamic configuration of ELEN and VLEN constant parameters in the RISCV Vector path. At this moment, if experimenting with different values of these parameters is needed, it is mandatory to recompile gem5 each time. 

I have been working on and testing a possible implementation during the last weeks. So, I could work together on this task if you agree with that.

Also, during my testing, I found some erroneous behaviours in specific instructions (i.e. vle instructions) when using a large enough VLEN value, generally above or equal to 2^14. This can be related to the microVl selected data type for implementing these instructions, although I am not 100% sure yet.

powerjg, 2023-08-07:
As far as I know, there is not anyone working on this. We would greatly appreciate the contribution!

powerjg, 2023-08-11:
In progress on #171

alvaromorenomv, 2023-10-20:
Finished on #171, I think we can close this one.