gem5/gem5 #165: GitHub Actions Runner VMs are limited to 60GB Disk Space. This means disabling PARSEC Tests

BobbyRBruce, 2023-08-07:
Our GitHub self-hosted Actions Runners run within VMs. These VMs are limited to a 60GB of disk image space. This is sufficient for most tests, but if running PARSEC, we are required to download the PARSEC Disk Image resource, v1.0.0 (http://resources.gem5.org/resources/x86-parsec?version=1.0.0). Unzipped this image is 33GB, which pushes the VM to exaust all it's disk image space. This crashes the VM.

Pull Request #164 disables the tests which use PARSEC.

The PARSEC tests can be re-incorporated when we are able to increase the disk image size on these VMs.

abmerop, 2023-08-15:
Trying to understand this more, does this mean that gem5, build, etc.  plus the disk image size must currently fit in 60GB?  I.e., what are the other 27GB (=60GB - 33GB) being used for?  I want to understand since a GPU full system test would need a disk of similar size (~31 GB), if one is ever added.

BobbyRBruce, 2023-08-16:
> Trying to understand this more, does this mean that gem5, build, etc. plus the disk image size must currently fit in 60GB? I.e., what are the other 27GB (=60GB - 33GB) being used for? I want to understand since a GPU full system test would need a disk of similar size (~31 GB), if one is ever added.

The downloaded zip of the image still exists as it's being unzipped. I think we're short of about 2GB when it's being decompressed and the VM fails. I think the zip is about 12GB. The remainder is the OS, gem5, and the gem5 build. So it's something like:

~33G : The Disk image uncompressed.
~12GB: The Disk image zipped.
~15GB: OS, GitHub Actions runner, gem5, and its dependencies.

closed 2023-11-14 by powerjg