gem5/gem5 #181: RISC-V full system simulation not working within systemc (gem5 within systemc)

voertler, 2023-08-14:
**Describe the bug**
Running a riscv-fs example within systemc using gem5 within systemc leads to a segmentation fault:

```./gem5.debug.sc m5out/config.ini 

        SystemC 2.3.4-Accellera --- Aug  4 2023 09:56:03
        Copyright (c) 1996-2022 by all Contributors,
        ALL RIGHTS RESERVED
Global frequency set at 1000000000000 ticks per second
============================================================
EventQueue Dump  (cycle 0)
------------------------------------------------------------
<No Events>
============================================================
build/ALL/sim/kernel_workload.cc:46: info: kernel located at: /root/.cache/gem5/riscv-bootloader-vmlinux-5.10

Info: board.platform.rtc: Real-time clock set to Sun Jan  1 00:00:00 2012
board.platform.terminal: Listening for connections on port 3456
build/ALL/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
build/ALL/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
board.remote_gdb: Listening for connections on port 7000
build/ALL/arch/riscv/linux/fs_workload.cc:51: info: Loading DTB file: m5out/device.dtb at address 0x87e00000
sc_module.cc:257: info: Entering event queue @ 0.  Starting simulation...
gem5 has encountered a segmentation fault!

--- BEGIN LIBC BACKTRACE ---
/root/mnt/gem5/build/ALL/libgem5_debug.so(_ZN4gem515print_backtraceEv+0x1f)[0x7f28a5a810f8]
/root/mnt/gem5/build/ALL/libgem5_debug.so(+0x65447ad)[0x7f28a5a9e7ad]
/lib64/libc.so.6(+0x4eb50)[0x7f289e67fb50]
/root/mnt/gem5/build/ALL/libgem5_debug.so(_ZN4gem514SignalSinkPortIbE3setERKb+0x1b)[0x7f28a51fe527]
/root/mnt/gem5/build/ALL/libgem5_debug.so(_ZN4gem516SignalSourcePortIbE3setERKb+0x35)[0x7f28a51fe0d5]
/root/mnt/gem5/build/ALL/libgem5_debug.so(_ZN4gem516IntSourcePinBase5raiseEv+0x23)[0x7f28a51fdd09]
/root/mnt/gem5/build/ALL/libgem5_debug.so(_ZN4gem58RiscvRTC3RTC11handleEventEv+0x75)[0x7f28a55ac6fb]
/root/mnt/gem5/build/ALL/libgem5_debug.so(_ZN4gem58MC1468188RTCEvent7processEv+0x108)[0x7f28a51a2858]
/root/mnt/gem5/build/ALL/libgem5_debug.so(_ZN4gem510EventQueue10serviceOneEv+0x10b)[0x7f28a5a92451]
./gem5.debug.sc[0x4294d9]
./gem5.debug.sc[0x44bec3]
./gem5.debug.sc[0x44cd01]
./gem5.debug.sc[0x44cf5d]
./gem5.debug.sc[0x41d9bd]
./gem5.debug.sc[0x42ff54]
/lib64/libc.so.6(__libc_start_main+0xe5)[0x7f289e66bd85]
./gem5.debug.sc[0x41ba1e]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 
```

When looking into the issue using gdb, this is the demangled stacktrace of the segfault:
```
#0  0x00007ffff5bc6527 in gem5::SignalSinkPort<bool>::set (this=0x0, new_state=@0x7fffffffdf8f: true) at build/ALL/sim/signal.hh:60
#1  0x00007ffff5bc60d5 in gem5::SignalSourcePort<bool>::set (this=0x14a8920, new_state=@0x7fffffffdf8f: true) at build/ALL/sim/signal.hh:108
#2  0x00007ffff5bc5d09 in gem5::IntSourcePinBase::raise (this=0x14a8920) at build/ALL/dev/intpin.hh:89
#3  0x00007ffff5f746fb in gem5::RiscvRTC::RTC::handleEvent (this=0x14a8450) at build/ALL/dev/riscv/rtc.cc:67
#4  0x00007ffff5b6a858 in gem5::MC146818::RTCEvent::process (this=0x14a8480) at build/ALL/dev/mc146818.cc:327
#5  0x00007ffff645a451 in gem5::EventQueue::serviceOne (this=0x9c75c0) at build/ALL/sim/eventq.cc:249
#6  0x00000000004294d9 in Gem5SystemC::Module::eventLoop (this=0x7fffffffe300) at sc_module.cc:241
#7  0x000000000044bec3 in sc_core::sc_simcontext::simulate(sc_core::sc_time const&) ()
#8  0x000000000044cd01 in sc_core::sc_start(sc_core::sc_time const&, sc_core::sc_starvation_policy) ()
#9  0x000000000044cf5d in sc_core::sc_start() ()
#10 0x000000000041d9bd in sc_main (argc=2, argv=0x9cfba0) at main.cc:402
#11 0x000000000042ff54 in sc_elab_and_sim ()
#12 0x00007fffef033d85 in __libc_start_main () from /lib64/libc.so.6
#13 0x000000000041ba1e in _start ()
```

This issue only occurs when using the gem5 within systemc coupling (util/systemc/gem5_within_systemc) or the SystemC TLM couplings. When running the examples just using the normal gem5 executable e.g. gem5.debug / gem5.opt it works as expected. 

The ARM gem5_library examples e.g. arm-ubuntu-run.py work as expected within systemc. Therefore I assume the error is somewhere within the MC146818 or RiscvRTC component, which are used in  RiscvBoard.

**Affects version**

gem5 23.0.0.1 , didn't try other versions

**gem5 Modifications**

No Modifications of gem5

**To Reproduce**

- Build gem5 and libgem5 e.g.

```
scons --with-cxx-config --without-tcmalloc USE_SYSTEMC=0 --duplicate-sources build/ALL/gem5.debug -j$(nproc)
scons --with-cxx-config --without-python --without-tcmalloc USE_SYSTEMC=0 --duplicate-sources build/ALL/libgem5_debug.so -j$(nproc)
```
- Build gem5_within_systemc (see util/systemc/gem5_within_systemc/README for details) i.e.
   - Build external systemc SystemC 
```
wget -q https://github.com/accellera-official/systemc/archive/refs/tags/2.3.4.tar.gz -O systemc-2.3.4.tar.gz \
    && tar -xf systemc-2.3.4.tar.gz
cd systemc-2.3.4
.config/bootstrap
mkdir build
cd build
../configure --prefix=/opt/systemc-2.3.4  --disable-shared --build=x86_64-pc-linux-gnu   \
    && make -j$(nproc) \
    && make install
```

 Adapt the Makefile in  util/systemc/gem5_within_systemc to point to systemc and do `make`

Now it is possible to compare gem5 with systemc to a simulation without system:
`
../../../build/ALL/gem5.debug ../../../configs/example/gem5_library/riscvmatched-fs.py 
` --> works as expected

Now run gem5 within systemc using the created config.ini
`
./gem5.debug.sc m5out/config.ini  
` --> leads to degmenation fault at the top

**Expected behavior**
No segmentation fault as when using the normal gem5 executable e.g. gem5.debug / gem5.opt

**Host Operating System**
Rocky 8.6

**Host ISA**
RISC-V

**Compiler used**
gcc (GCC) 8.5.0 20210514 (Red Hat 8.5.0-18)

powerjg, 2023-08-14:
Thanks for the detailed bug report!

My bet is that the problem is that in the C++ config initialization code we're not handling the SignalPorts correctly. IIRC, there's special code to connect up ports, but when someone converted the pins to signals using the port interface they probably did not also fix the C++ initialization code.

Let us know if you find a solution. Otherwise, we'll add this to the backlog :)

voertler, 2023-08-14:
Thanks for the fast reply! I was suspecting some not correctly bound ports/signals as the target of the set method is null. Do you have any hint where is a good starting point for looking at the C++ initialisation code?

powerjg, 2023-08-14:
I would start looking around here: https://github.com/gem5/gem5/blob/develop/src/sim/cxx_manager.cc#L522

You can use `--debug-flags=CxxConfig` to look into what is being done. I would guess that it's not calling `bind` on the signal like it should (see https://github.com/gem5/gem5/blob/develop/src/sim/cxx_manager.cc#L482 and https://github.com/gem5/gem5/blob/develop/src/sim/port.hh#L118). You need to call `bind(peer)` on either the source or the sink port.

voertler, 2023-08-16:
After some debugging the cause seems to be within: :https://github.com/gem5/gem5/blob/f6d44ac7b3e7def977f99ab542cd10118d3b6876/build_tools/cxx_config_cc.py#L107 

The RiscVRTC component uses an IntSourcePin, which defines as port.role "Int Source Pin". Therefore the wrong RiscvRTC.cc gets generated. 

The check `is_requestor = port.role == "GEM5 REQUESTOR"` seems outdated. Shouldn't the Port.is_source member be uesed to identify a requestor?

powerjg, 2023-08-16:
I think you're correct! I would be good with that change. Would you mind making a PR?

BobbyRBruce, 2023-08-21:
Fixed with #196

closed 2023-08-21 by BobbyRBruce