gem5/gem5 #192: Bug with workloads and suggested change to JSON schema.

Harshil2107, 2023-08-17:
The current workloads lack the specification of resource versions for the resources they contain. This issue arises when attempting to retrieve a specified resource from the workload. In this scenario, the system automatically resorts to using the latest resource version that is compatible with the installed gem5 version. However, this behavior is not the intended one. Ideally, a workload should explicitly define the precise version of the resource it requires.

This situation can also lead to complications when multiple data sources contain resources required by a workload. Instead of adhering to the resource version specified in the workload, the "obtain resource" function retrieves the latest compatible version from all available data sources. This behavior diverges from the intended outcome.

The current workload structure is as follows:

```
{
    "category": "workload",
    "id": "x86-matrix-multiply-omp-100-8-looppoint-region-7",
    "description": "A workload which loads a Looppoint JSON file for the 'x86-matrix-multiply-omp' binary with inputs '100' and '8' and runs region 7 via a checkpoint.",
    "function": "set_se_looppoint_workload",
    "resources": {
        "binary": "x86-matrix-multiply-omp",
        "looppoint": "x86-matrix-multiply-omp-100-8-looppoint",
        "checkpoint": "x86-matrix-multiply-omp-100-8-looppoint-checkpoint-region-7"
    },
    "additional_params": {
        "region_id": "7"
    },
    "architecture": "X86",
    "size": 0,
    "tags": [],
    "code_examples": [],
    "license": "",
    "author": [],
    "source_url": "",
    "resource_version": "1.0.0",
    "gem5_versions": [
        "23.0"
    ],
    "example_usage": "Workload(\"x86-matrix-multiply-omp-100-8-looppoint-region-7\")"
}
```

The new change would be to change the `resources` field to `obtain_resources`. The `obtain_resources` field would be a dictionary that has a category of resource for example `"binary"` as key and a dictionary containing resource id and version as value.

The above resource will look as follows with the new changes:
```
{
    "category": "workload",
    "id": "x86-matrix-multiply-omp-100-8-looppoint-region-7",
    "description": "A workload which loads a Looppoint JSON file for the 'x86-matrix-multiply-omp' binary with inputs '100' and '8' and runs region 7 via a checkpoint.",
    "function": "set_se_looppoint_workload",
    "obtain_resources": {
        "binary": {
             "id": "x86-matrix-multiply-omp",
             "resource_version": "1.0.0"
         }
        "looppoint":{
             "id": "x86-matrix-multiply-omp-100-8-looppoint",
             "resource_version": "1.0.0"
         }
        "checkpoint": {
             "id": "x86-matrix-multiply-omp-100-8-looppoint-checkpoint-region-7",
             "resource_version": "1.0.0"
         }
    },
    "additional_params": {
        "region_id": "7"
    },
    "architecture": "X86",
    "size": 0,
    "tags": [],
    "code_examples": [],
    "license": "",
    "author": [],
    "source_url": "",
    "resource_version": "2.0.0",
    "gem5_versions": [
        "23.1"
    ],
    "example_usage": "Workload(\"x86-matrix-multiply-omp-100-8-looppoint-region-7\")"
}
```
This change will add a new version for all resources and this new version will be compatible with gem5 version 23.1 and onwards.

Incorporating this new change would necessitate an update to the resources schema. As modifications accumulate within resources, maintaining a universally applicable schema becomes progressively intricate. To address this challenge, a pragmatic approach could involve maintaining distinct schemas tailored to different gem5 versions. Subsequently, a validation script could be developed, leveraging the `gem5_versions` attribute of each resource to ensure alignment with the appropriate schema. This approach would contribute to a more streamlined and manageable system as gem5 evolves.

Harshil2107, 2023-08-17:
Todos:

- [ ] Figure out where to store schemas for older versions, and how will they be accessed.

- [ ] Figure out how validation of multiple schemas works with MongoDB.

- [ ] Create an updated schema for the new gem5 version.

- [ ] Add workload changes to the schema.

- [ ] Write scripts that can update all existing workloads to new versions following any changes in dependent resources or an update to the schema.

- [ ] Update all workloads to a new version following the schema change.

- [ ] Update stdlib implementation of workload to reflect the changes.

BobbyRBruce, 2023-08-18:
Going to add some notes in here:

> Todos:
>
> * [ ]  Figure out where to store schemas for older versions, and how will they be accessed.
> * [ ]  Figure out how validation of multiple schemas works with MongoDB.
> * [ ]  Create an updated [...]

I don't understand this one. Do you mean create new versions of existing workloads to conform to the new schema design?

> * [ ]  Update all workloads to a new version following the schema change.

This is essentially part of the "Write scripts that can update all existing workloads " task, right? These new workloads should be for v23.1, right?

What I _do_ think needs to be an explicit step is the the updating of the non-Workload resources to state they are compatible with v23.1

> * [ ]  Update stdlib implementation of workload to reflect the changes.

I think this step needs to be done at the same time as we add the new data conforming to the new schema. If we uddate gem5 before then it'll try to access resources it doesn't have compatibility with, but if we update the data first the gem5 binary can't parse the data in the new format... it's a bit annoying. There me be an awkward overlap here when the data is updated, breaking develop for people, and then you get your PR merged.

Harshil2107, 2023-08-18:
> I don't understand this one. Do you mean create new versions of existing workloads to conform to the new schema design?

Yes, but also the script that updates a workload version when a dependent resource is updated.

> This is essentially part of the "Write scripts that can update all existing workloads " task, right? These new workloads should be for v23.1, right?

Yes, but I separated them because it seemed like two different steps, one is to create scripts and then update.

> What I do think needs to be an explicit step is the updating of the non-Workload resources to state they are compatible with v23.1

Do you mean that when a workload is updated to be compatible with the new gem5 version, it updates its dependent resources as well? 

If the same resource and version are compatible with the new gem5 version we just append to the list of `gem5_versions` in JSON, so should that also happen recursively, that is if a new gem5 version is added to a resource then it also gets added to the corresponding workload? If that is the case then we would need to directly write to the database.

BobbyRBruce, 2023-08-18:
This issue does somewhat relate to Discussion #201. Please engage in this discussion to ensure we do this right.