gem5/gem5 #197: Bug when trying to restore checkpoints in SPARC: “panic: panic condition !pte occurred: Tried to execute unmapped address 0.”

Harshil2107, 2023-08-18:
**Describe the bug**
gem5 throws a `panic: panic condition !pte occurred: Tried to execute unmapped address 0.` when we try to restore a saved checkpoint in SPARC isa. The saved checkpoint is of running hello world binary using a simple board. 

**Changes made**
Added checkpoint tests for various isas in #167.

**To Reproduce**
Note: If #167 is not merged yet, please use the `sparc-hello-save-checkpoint.py` and `sparc-hello-restore-checkpoint.py` added in the pull request.
Steps to reproduce the behavior. Please assume starting from a clean repository:
1. Build gem5 with ALL: `scons build/ALL/gem5.opt -j(number of threads)`
2. Run the sparc-hello-save-checkpoint.py file with the following command:
```./build/ALL/gem5.opt tests/gem5/checkpoint_tests/configs/sparc-hello-save-checkpoint.py```
3. Make sure that a directory named “sparc-hello-test-checkpoint” is created in the current working directory.
4. Run the sparc-hello-restore-checkpoint.py file with the following command:
```./build/ALL/gem5.opt tests/gem5/checkpoint_tests/configs/sparc-hello-restore-checkpoint.py```

**Terminal Output**
```
./build/ALL/gem5.opt tests/gem5/checkpoint_tests/configs/sparc-hello-restore-checkpoint.py 
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version DEVELOP-FOR-23.1
gem5 compiled Aug 18 2023 09:33:03
gem5 started Aug 18 2023 09:41:53
gem5 executing on challenger, pid 2026302
command line: ./build/ALL/gem5.opt tests/gem5/checkpoint_tests/configs/sparc-hello-restore-checkpoint.py

info: Using default config
Global frequency set at 1000000000000 ticks per second
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (32 Mbytes)
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
board.remote_gdb: Listening for connections on port 7001
src/sim/process.cc:396: warn: Checkpoints for pipes, device drivers and sockets do not work.
src/sim/simulate.cc:194: info: Entering event queue @ 10104885.  Starting simulation...
src/arch/sparc/faults.cc:630: panic: panic condition !pte occurred: Tried to execute unmapped address 0.
Memory Usage: 218712 KBytes
Program aborted at tick 30808161
--- BEGIN LIBC BACKTRACE ---
./build/ALL/gem5.opt(_ZN4gem515print_backtraceEv+0x30)[0x56383860ce30]
./build/ALL/gem5.opt(_ZN4gem512abortHandlerEi+0x4e)[0x56383863200e]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x14420)[0x7f43b33a1420]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb)[0x7f43b256d00b]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b)[0x7f43b254c859]
./build/ALL/gem5.opt(+0x1b2ca65)[0x563837a7aa65]
./build/ALL/gem5.opt(_ZN4gem58SparcISA28FastInstructionAccessMMUMiss6invokeEPNS_13ThreadContextERKNS_14RefCountingPtrINS_10StaticInstEEE+0x1f5)[0x56383a787715]
./build/ALL/gem5.opt(_ZN4gem513BaseSimpleCPU9advancePCERKSt10shared_ptrINS_9FaultBaseEE+0xb4)[0x5638384b59e4]
./build/ALL/gem5.opt(_ZN4gem515TimingSimpleCPU11advanceInstERKSt10shared_ptrINS_9FaultBaseEE+0x88)[0x5638384ad618]
./build/ALL/gem5.opt(_ZN4gem515TimingSimpleCPU9sendFetchERKSt10shared_ptrINS_9FaultBaseEERKS1_INS_7RequestEEPNS_13ThreadContextE+0x5b)[0x5638384adcab]
./build/ALL/gem5.opt(_ZN4gem58SparcISA3TLB15translateTimingERKSt10shared_ptrINS_7RequestEEPNS_13ThreadContextEPNS_7BaseMMU11TranslationENS9_4ModeE+0x87)[0x56383a7d0817]
./build/ALL/gem5.opt(_ZN4gem515TimingSimpleCPU5fetchEv+0x2ad)[0x5638384aeaed]
./build/ALL/gem5.opt(_ZN4gem510EventQueue10serviceOneEv+0xc2)[0x5638386210a2]
./build/ALL/gem5.opt(_ZN4gem59doSimLoopEPNS_10EventQueueE+0x70)[0x563838649fd0]
./build/ALL/gem5.opt(_ZN4gem58simulateEm+0x293)[0x56383864a653]
./build/ALL/gem5.opt(+0x2663d62)[0x5638385b1d62]
./build/ALL/gem5.opt(+0x1b13bf6)[0x563837a61bf6]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x2a8748)[0x7f43b365a748]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x8dd8)[0x7f43b342ff48]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x8fb)[0x7f43b357ce4b]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyFunction_Vectorcall+0x94)[0x7f43b365a124]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74d6d)[0x7f43b3426d6d]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x7d86)[0x7f43b342eef6]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x8fb)[0x7f43b357ce4b]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyFunction_Vectorcall+0x94)[0x7f43b365a124]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74d6d)[0x7f43b3426d6d]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0xea8)[0x7f43b3428018]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x8fb)[0x7f43b357ce4b]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyEval_EvalCodeEx+0x42)[0x7f43b357d1d2]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyEval_EvalCode+0x1f)[0x7f43b357d5bf]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x1cfc01)[0x7f43b3581c01]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x25f547)[0x7f43b3611547]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 

Aborted (core dumped)
```

ivanaamit, 2024-01-08:
Hi @shingarov, I've noticed that you've self-assigned this issue. Are you currently working on it? Thanks.

erin-le, 2024-11-12:
Reproduced this error for gem5 v24.0.0.1, develop ad8bd6b5c7a6f72060d4079604a8a9a8f61df9e6