gem5/gem5 #213: Workload append json doesn't work as expected

powerjg, 2023-08-23:
I am trying to extend gem5 resources with my own specialized workloads and running into problems. I'm not totally sure what's going on, and I haven't dug very deep on it.

**Affects version**
23.0

**gem5 Modifications**
No modifications.

I have created a new file `npb-workloads.json` by running the following python code.

```python
template = """
        {{
            "id": "x86-npb-{benchmark}-{size}",
            "description": "NAS Parallel benchmarks - {Benchmark} - Class {Size}.",
            "function": "set_kernel_disk_workload",
            "resources": {{
                "kernel": "x86-linux-kernel-4.19.83",
                "disk_image": "x86-npb",
                "readfile_contents": "/home/gem5/NPB3.3-OMP/bin/{benchmark}.{Size}.x; sleep 5; m5 exit;"
            }},
            "additional_params": {{}},
            "architecture": "X86",
            "resource_version": "1.0.0",
            "gem5_versions": [
                "23.0"
            ]
        }},"""

res = """
{
    "resources": [
"""

for benchmark in ["bt", "cg", "ep", "ft", "is", "lu", "mg", "sp"]:
    for size in ["A", "B", "C", "D"]:
        res += template.format(benchmark=benchmark, Benchmark=benchmark.upper(), Size=size, size=size.lower())

res += """
    ]
}
"""

with open("npb-workloads.json", "w") as f:
    f.write(res)
```

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:
1. Compile gem5 with command ` scons build/ALL/gem5.opt`
2. Modify `configs/example/gem5_library/x86-npb-benchmarks.py` to use `board.set_kernel_disk_workload(obtain_resource("x86-npb-cg-a"))`
3. Execute the simulation with ` GEM5_RESOURCE_JSON_APPEND="npb-workloads.json" build/ALL/gem5.opt configs/example/gem5_library/x86-npb-benchmarks.py --benchmark cg --size A`

**Terminal Output**
```
info: Using default config
info: Appending resources from npb-workloads.json
warn: Error getting resources from client GEM5_RESOURCE_JSON_APPEND: string indices must be integers
Exception: Resource with ID 'x86-npb-cg-a' not found.

At:
  src/python/gem5/resources/client_api/client_wrapper.py(151): get_resource_json_obj_from_client
  src/python/gem5/resources/client.py(152): get_resource_json_obj
  src/python/gem5/resources/resource.py(590): obtain_resource
  configs/example/gem5_library/x86-npb-benchmarks.py(204): <module>
  src/python/m5/main.py(630): main
```

**Expected behavior**
gem5 should download the npb resources and run CG

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

BobbyRBruce, 2023-08-23:
Small correction: I don't think if using `Workload` you need to specify category (though you will when we have Issue #212 implemented and get Workload via `obtain_resource`).

BobbyRBruce, 2023-08-23:
Closing this issue. No action needs taken.

closed 2023-08-23 by BobbyRBruce