gem5/gem5 #214: Documentation on creating local resources is lacking

BobbyRBruce, 2023-08-23:
Issue #213 highlighted that using and declaring local resources is difficult and there doesn't seem to be any documentation or prompts to guide a user. The following information should be communicated:

1. The format of a JSON file that will be parsed correctly. Right now unhelpful errors are given (tangentially, perhaps we need better error handling here-- an error should link back to this documentation).
2. The `GEM5_RESOURCE_JSON_APPEND` and `GEM5_RESOURCE_JSON` environment variables aren't documented anywhere.
3. Related to this is PR #204 (allowing users to specify local files, rather than just URLs for resource locations). This too should be documented.
4. Schema validation? Is this possible?
5. Documentation should provide some cookbook examples.

Putting this issue here, but perhaps best place for the bulk of this documentation is the gem5 website: https://github.com/gem5/website.

BobbyRBruce, 2023-08-23:
Note: PR https://github.com/gem5/gem5-resources/pull/2 provides a good start. I think this should maybe be put on the website as a grander documentation of Resources and how to use them.

powerjg, 2023-08-23:
I'll add that we also need to make the error information better. Saying "can't use string as integer index" or something like that was not helpful in tracking down the problem.

kunpai, 2023-08-23:
I like the idea of adding documentation of local Resources to the website.

I think number 1 and number 4 are related, i.e., a more helpful error message if the JSON entered by the user is incorrect could be solved through schema validation. Personally, it has been a while since I worked with a schema validation library, so I am not sure, but I believe it is possible.

While I also agree that the gem5 website is the best place to host this documentation, it should be linked on the gem5 Resources website [http://resources.gem5.org](http://resources.gem5.org) as well.

Harshil2107, 2023-08-23:
For number 4, as we cannot use packages, we would have to manually validate the schema. 

An alternative solution:  On the tool that I am working on there is a command to validate a JSON file against the schema, maybe if there is an error for a missing field, we can say to use the tool and make sure that the JSON conforms to the schema?

BobbyRBruce, 2023-08-23:
He @kunpai : Is it ok if i  take your work and incorporate it into a webpage with expanded documentation on the points addressed?

kunpai, 2023-08-23:
@BobbyRBruce that is fine by me. Do let me know how I can help with this documentation as well 😀

Harshil2107, 2024-01-11:
These are the PRs that are related to this issue:
https://github.com/gem5/website/pull/90
https://github.com/gem5/website/pull/82

closed 2024-03-07 by ivanaamit