gem5/gem5 #226: CLFLUSH/CLFLUSHOPT/CLWB causes segmentation faults

AKKamath, 2023-08-24:
**Describe the bug**
Executing these operations on write-protected pages causes segmentation faults in GEM5. 
For example the below code causes segmentation faults in GEM5:
```cpp
#include <fcntl.h>
#include <stdio.h>
#include <stdlib.h>
#include <sys/mman.h>
#include <sys/stat.h>
#include <unistd.h>
#include <x86intrin.h>

int main(int argc, char *argv[])
{
    char *addr = (char *)mmap(NULL, 4096, PROT_READ, MAP_PRIVATE | MAP_ANONYMOUS, -1, 0);
    _mm_clwb(addr);
    munmap(addr, 4096);
    return 0;
}
```
According to the ISA documentation:
> The CLWB instruction can be used at all privilege levels and is subject to all permission checking and faults associated with a byte load.

I've confirmed this error occurs on TimingSimpleCPU and O3CPU; it does not occur for X86KVMCPU.

A possible solution for O3CPU could be to change this line:
https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/src/cpu/o3/lsq.cc#L1154

```cpp
void
LSQ::LSQRequest::sendFragmentToTranslation(int i)
{
    numInTranslationFragments++;
    _port.getMMUPtr()->translateTiming(req(i), _inst->thread->getTC(),
            this, isLoad() ? BaseMMU::Read : BaseMMU::Write);
}
```
to
```cpp
void
LSQ::LSQRequest::sendFragmentToTranslation(int i)
{
    numInTranslationFragments++;
    _port.getMMUPtr()->translateTiming(req(i), _inst->thread->getTC(),
            this, isLoad() || req(i)->isCacheClean() ? BaseMMU::Read : BaseMMU::Write);
}
```

**Affects version**
Current stable version. Commit # f29bfc0640c88a79eb7f94454ce31b3237ec0066

BobbyRBruce, 2023-08-24:
@mkjost0 : Can you:

1. rRplicate this for us? If you could include the gem5 configuration script you created to reproduce this, and, just for good house keeping the command you used to compile the binary.
2. Apply the O3CPU change suggested and verify it works (at least, in the sence it doesn't produce a segfault.
3. Assuming the O3CPU change "works", push as PR and link here.

Note: We'll need to figure out how to fix the issue with TimingSimpleCPU.

AKKamath, 2023-08-25:
On double-checking, I don't think my previous suggestion was the best. This issue is only for X86 ISA, and the code changes I suggested may have unintended consequences for other ISAs.

Perhaps changes inside the X86 TLB translate code instead? Specifically lines:
https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/src/arch/x86/tlb.cc#L511
https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/src/arch/x86/tlb.cc#L519
https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/src/arch/x86/tlb.cc#L554

Adding the following lines there should help:
```cpp
// CLFLUSHOPT/WB/FLUSH should be treated as read
if(req->isCacheClean())
    mode = BaseMMU::Read;
```

As this change is within X86 specific code, it shouldn't have side effects on non-X86 ISAs.

powerjg, 2023-09-05:
I think this is actually a deeper issue than either of these solutions address.

I think that the `CLFLUSH`/`CLFLUSHOPT`/`CLWB` instructions in x86 don't map well to their counterparts in Arm. In gem5, the request flags were created for the Arm instructions, not x86.

~~So, I think we need to change the way these x86 instructions are implemented and possibly extend request to have the right flags.~~

Ok, so after thinking about this more, I guess the problem is that in x86 these instructions should be able to proceed to a read-only page, but they should appear as stores in the pipeline (i.e., they must respect the `sfence` instruction and they are committed like stores). 

So, I think the best solution would be to check if the request `isCacheClean` in the TLB as you suggest. However, I'm concerned about changing the `mode`. Instead, we should change the permission check where we check for read-only pages and add the call to `req->isCacheClean()` there.

E.g., here: https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/src/arch/x86/tlb.cc#L367

@AKKamath, do you want to take a stab at a PR?

AKKamath, 2023-09-15:
While your suggestion does fix this for TLB checks (with a modification to [line 470](https://github.com/gem5/gem5/blob/f29bfc0640c88a79eb7f94454ce31b3237ec0066/src/arch/x86/tlb.cc#L470) as well), the issue still exists on page faults.

If the translation is missing in the TLB, the page walk is started. When the page walk is completed, if the page is missing in the table, GEM5 sends a page fault interrupt to the OS (in my case, Linux) along with the details of the access. There, Linux allocates a page, gets the page protection details from the Page Table Entry and checks it against the access details that GEM5 shared. When it sees that the access was a write to a read-only page, it sends a segmentation fault.

My previous suggestion should avoid this as it starts treating the access as a read early on during the translation process, so that on a page fault Linux is told that the access was a read.

AKKamath, 2023-11-23:
I've created a PR at #592 with the changes I mentioned.

BobbyRBruce, 2023-11-29:
Fixed by #592

closed 2023-11-29 by BobbyRBruce

hnpl, 2024-02-09:
The #592 change caused the following assertion error on my setup when executing a CLFLUSH instruction.

```
src/sim/simulate.cc:199: info: Entering event queue @ 14392355269750.  Starting simulation...                                                                                            
src/sim/simulate.cc:199: info: Entering event queue @ 15473386459500.  Starting simulation...                                                                      
src/sim/power_state.cc:105: warn: PowerState: Already in the requested power state, request ignored                                                                
src/sim/simulate.cc:199: info: Entering event queue @ 15474400558750.  Starting simulation...                                                                      
src/mem/ruby/system/Sequencer.cc:668: warn: Replacement policy updates recently became the responsibility of SLICC state machines. Make sure to setMRU() near callba
cks in .sm files!                        
gem5.opt: src/cpu/translation.hh:257: void gem5::DataTranslation<ExecContextPtr>::finish(const Fault&, const RequestPtr&, gem5::ThreadContext*, gem5::BaseMMU::Mode)
 [with ExecContextPtr = gem5::TimingSimpleCPU*; gem5::Fault = std::shared_ptr<gem5::FaultBase>; gem5::RequestPtr = std::shared_ptr<gem5::Request>]: Assertion `mode
== state->mode' failed.                  
Program aborted at tick 15478398983501
```