gem5/gem5 #252: scons: Enable to pass the daily CI and weekly CI after kconfig build system merged

rogerchang23424, 2023-09-01:
The comment posted by @BobbyRBruce: https://github.com/gem5/gem5/issues/69#issuecomment-1644441362_

> I'd like to raise the potential issue that this may break some of our testing infrastructure. The tests run via "tests/main.py", which I call "testlib", will build the gem5 binaries they need to run the tests and I think some of the changes outlined here might break this.

> The code for creating a gem5 binary for a test is here: https://github.com/gem5/gem5/blob/af72b9ba580546ac12ce05bfaac3fd53fa8699f4/tests/gem5/fixture.py#L186. As you can see it constructs the build command by utilizing the `PROTOCOL` variable and depeneds on the files "build_opts".

> While most of our tests build a pretty vanilla "build/ALL/gem5.opt" binary to run tests, other target specific ISAs with specific protocols.

> I think `./main.py list --length=long gem5/riscv-boot-tests` should sanity test any changes you have. Tests invoked with that command try to "build/NULL_MI_example/gem5.opt".

After the gem5 change to kconfig build system #69, The customized option setting `scons PROTOCOL=xxx gem5/ARM/gem5.opt` and `--default` options will not available anymore. Create the issue to tracking the support of customized options with kconfig build system

rogerchang23424, 2023-09-20:
Hi,
I would like to know how to run GitHub CI in the local machine

powerjg, 2023-09-21:
There's not a great way to run github actions locally. The best we can find is shown in this (closed) PR: https://github.com/gem5/gem5/pull/176

rogerchang23424, 2023-12-01:
The daily tests and weekly tests work well, close the issue.

closed 2023-12-01 by rogerchang23424