gem5/gem5 #253: Daily long GPU tests failing due to `(m_atomicLog.size() > 0)` assertion when running random tester

BobbyRBruce, 2023-09-01:
**Describe the bug**

The Daily Long GPU Tests are failing: https://github.com/gem5/gem5/actions/runs/6034233301/job/16377497140.

An assertion is being triggered when running the "configs/example/ruby_gpu_random_test.py" script with `VEGA_X86/gem5`:

`Assertion failed: (m_atomicLog.size() > 0), function popAtomicLogEntryFront, file DataBlock.cc, line 141.`

I have managed to reproduce this bug locally.

**Affects version**

The bug exists, at the time of writing, on the [develop branch of gem5](https://github.com/gem5/gem5/tree/develop).

Known to not affect revision  [c156df6](https://github.com/gem5/gem5/commit/c156df620de59b1379f2ac6e64ec25cfdd7d1a62) due to success of [Daily tests running on that revision](https://github.com/gem5/gem5/actions/runs/6021764392)

First observed in [Daily Test failure](https://github.com/gem5/gem5/actions/runs/6034233301) at revision [0e323bc4](https://github.com/gem5/gem5/commit/0e323bc40953670430cc9185d1a51625ee4b43ee).

Reproduced locally at [4de4e22](https://github.com/gem5/gem5/commit/4de4e2255306e1b784080854cb9a2150c8c09a65).

**Possible breaking patches**

Using `git log --oneline c156df6..0e323bc4 `, I can determine the following PRs are the only candidates PRs which could break these tests:

* 0e323bc409 mem: Atomic ops to same address (#200)
* fceb7e05a3 util-docker: Add GitHub Action to create Docker Images (#236)

It seems very likely #200 is causing this.

**To Reproduce**
```sh
# Assume our pwd is the root of the gem5 repository,
# in-sync with https://github.com/gem5/gem5.

git switch develop
build/VEGA_X86/gem5.opt -j <num compilation threads>

./build/VEGA_X86/gem5.opt configs/example/ruby_gpu_random_test.py --test-length 50000 --num-dmas 0
```

**Terminal Output**

```sh
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version DEVELOP-FOR-23.1
gem5 compiled Sep  1 2023 00:28:04
gem5 started Sep  1 2023 00:32:08
gem5 executing on Bobbys-M2-Macbook-Pro.local, pid 89646
command line: ./build/VEGA_X86/gem5.opt configs/example/ruby_gpu_random_test.py --test-length 50000 --num-dmas 0

Attaching ruby ports to the tester
Creating 0 DMAs
Creating 1 WFs attached to 1 CUs
Global frequency set at 1000000000 ticks per second
warn: system.ruby.network adopting orphan SimObject param 'ext_links'
warn: system.ruby.network adopting orphan SimObject param 'int_links'
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
warn: rounding error > tolerance
    1.250000 rounded to 1
warn: rounding error > tolerance
    1.250000 rounded to 1
src/mem/dram_interface.cc:692: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (1024 Mbytes)
src/cpu/testers/gpu_ruby_test/protocol_tester.cc:140: warn: Data race check is not enabled
src/cpu/testers/gpu_ruby_test/protocol_tester.cc:157: warn: If `random_seed == 0` (or `random_seed` is unset) ProtocolTester does not seed the RNG. This will NOT result in the RNG generating different results each run. In this case the RNG is seeded by a default value. This differs from behavior in previous versions of gem5. Setting `random_seed` to a non-zero value is strongly recommended.
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
      1: global: Episode 0
Assertion failed: (m_atomicLog.size() > 0), function popAtomicLogEntryFront, file DataBlock.cc, line 141.
Program aborted at tick 521
--- BEGIN LIBC BACKTRACE ---
0   gem5.opt                            0x00000001013166a8 _ZN4gem515print_backtraceEv + 44
1   gem5.opt                            0x0000000101329da0 _ZN4gem512abortHandlerEi + 132
2   libsystem_platform.dylib            0x000000018eb42a84 _sigtramp + 56
3   libsystem_pthread.dylib             0x000000018eb13c28 pthread_kill + 288
4   libsystem_c.dylib                   0x000000018ea21ae8 abort + 180
5   libsystem_c.dylib                   0x000000018ea20e44 err + 0
6   gem5.opt                            0x000000010189a888 _ZNK4gem54ruby9DataBlock7getDataEii.cold.1 + 0
7   gem5.opt                            0x00000001017ce800 _ZN4gem54ruby9DataBlock21clearAtomicLogEntriesEv + 0
8   gem5.opt                            0x00000001017a3dd0 _ZN4gem54ruby12GPUCoalescer11hitCallbackEPNS0_16CoalescedRequestENS0_11MachineTypeERNS0_9DataBlockEbNS_6CyclesES7_S7_b + 624
9   gem5.opt                            0x00000001017a39e8 _ZN4gem54ruby12GPUCoalescer13writeCallbackEyNS0_11MachineTypeERNS0_9DataBlockENS_6CyclesES5_S5_b + 232
10  gem5.opt                            0x00000001017721e8 _ZN4gem54ruby14TCP_Controller14wab_bypassDoneERPNS0_7TCP_TBEERPNS0_9TCP_EntryEy + 172
11  gem5.opt                            0x000000010177f990 _ZN4gem54ruby14TCP_Controller18doTransitionWorkerENS0_9TCP_EventENS0_9TCP_StateERS3_RPNS0_7TCP_TBEERPNS0_9TCP_EntryEy + 2900
12  gem5.opt                            0x000000010177e338 _ZN4gem54ruby14TCP_Controller12doTransitionENS0_9TCP_EventEPNS0_9TCP_EntryEPNS0_7TCP_TBEEy + 112
13  gem5.opt                            0x0000000101782b88 _ZN4gem54ruby14TCP_Controller6wakeupEv + 936
14  gem5.opt                            0x00000001017ce040 _ZN4gem54ruby8Consumer19processCurrentEventEv + 272
15  gem5.opt                            0x0000000101321dc4 _ZN4gem510EventQueue10serviceOneEv + 244
16  gem5.opt                            0x000000010133c3ec _ZN4gem59doSimLoopEPNS_10EventQueueE + 272
17  gem5.opt                            0x000000010133be3c _ZN4gem58simulateEy + 424
18  gem5.opt                            0x00000001006ab4b0 _ZZN8pybind1112cpp_function10initializeIRPFPN4gem522GlobalSimLoopExitEventEyES4_JyEJNS_4nameENS_5scopeENS_7siblingENS_5arg_vEEEEvOT_PFT0_DpT1_EDpRKT2_ENKUlRNS_6detail13function_callEE_clESP_ + 84
19  gem5.opt                            0x0000000100653284 _ZN8pybind1112cpp_function10dispatcherEP7_objectS2_S2_ + 3968
20  Python                              0x000000010438a604 cfunction_call + 60
21  Python                              0x0000000104343aa8 _PyObject_Call + 124
22  Python                              0x000000010441cbf8 _PyEval_EvalFrameDefault + 48384
23  Python                              0x00000001044103b4 PyEval_EvalCode + 168
24  Python                              0x000000010440c798 builtin_exec + 332
25  Python                              0x000000010438b058 cfunction_vectorcall_FASTCALL_KEYWORDS + 76
26  Python                              0x000000010441acfc _PyEval_EvalFrameDefault + 40452
27  Python                              0x000000010441f480 _PyEval_Vector + 116
28  gem5.opt                            0x000000010069a528 _ZNK8pybind116detail10object_apiINS0_8accessorINS0_17accessor_policies8str_attrEEEEclILNS_19return_value_policyE1EJEEENS_6objectEDpOT0_ + 76
29  gem5.opt                            0x000000010132a80c main + 320
30  dyld                                0x000000018e7bbf28 start + 2236
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 
```
**Host Operating System**

Reproduced on a Macbook M2 System. Originally observed in Daily Tests, an ubuntu system.

**Host ISA**

ARM (Apple Silicon), also observed in X86

**Compiler used**

Clang 14.0.3, but also obsered in GCC

BobbyRBruce, 2023-09-01:
@mattsinc @abmerop @bujset : As mentioned in the post this appears to have been caused by #200. Could someone on your end try to come up with a fix for this?

abmerop, 2023-09-01:
#255 probably fixes this

BobbyRBruce, 2023-09-01:
> #255 probably fixes this

Thanks @abmerop! Let's get this in ASAP and verify.

PR #255 (2023-09-02): mem-ruby: Reorder SLC atomic and response actions
Currently the MOESI_AMD_Base-directory transition for system level atomics sends the response message before the atomic is performed. This was likely done because atomics are supposed to return the value of the data *before* the atomic is performed and by simply ordering the actions this way that was taken care of.

With the new atomic log feature, the atomic values are pulled from the log by the coalescer on the return path. Therefore, these actions can be reordered. In fact, it is now necessary that the atomics be performed before sending the [... cut]
commit 2da54d5a (2023-09-02): mem-ruby: Reorder SLC atomic and response actions

Currently the MOESI_AMD_Base-directory transition for system level
atomics sends the response message before the atomic is performed. This
was likely done because atomics are supposed to return the value of the
data *before* the atomic is performed and by simply ordering the actions
this way that was taken care of.

With the new atomic log feature, the atomic values are pulled from the
log by the coalescer on the return path. Therefore, these actions can be
reordered. However, it is now necessary that the atomics be performed
before sending th [... cut]
commit 2eeecc53 (2023-09-02): mem-ruby: Reorder SLC atomic and response actions (#255)

Currently the MOESI_AMD_Base-directory transition for system level
atomics sends the response message before the atomic is performed. This
was likely done because atomics are supposed to return the value of the
data *before* the atomic is performed and by simply ordering the actions
this way that was taken care of.

With the new atomic log feature, the atomic values are pulled from the
log by the coalescer on the return path. Therefore, these actions can be
reordered. In fact, it is now necessary that the atomics be performed
before sen [... cut]
closed 2023-09-08 by Harshil2107