gem5/gem5 #260: Simulation entering deadlock when using X86KvmCPU on a i7-13700 host

BobbyRBruce, 2023-09-01:
This was reported to me by a gem5 user in our Slack channel. Both the user and I find this bug to be rather strange, and we're still unsure as to what the root cause is. However, all other reasonable assumptions were ruled out and we could only conclude that the i7-13700 processor, or perhaps something in its bios settings, was the cause.

I am logging this issue to see if anyone else ever encounters it and can perhaps give more information.

The i7-13700 processor was only released in January of 2023, which makes me think that if it _is_ an issue with the processor it may not be surprising more people haven't encountered it yet. 

**Bug Description**

The user observed that `./build/X86/gem5.opt gem5/configs/example/gem5_library/x86-ubuntu-run-with-kvm.py` reaches a deadlock during simulate but only after moving from a i7-8700/Ubuntu 20.04/Linux 5.4 system, where this did not occur, to a i7-13700/Ubuntu 22.04/Linux 6.2 system where it did.

The i7-13700 system's terminal output appears to hang indefinitely after a certain point (this is shown below, but it is not particularly useful information).

The user suspects it was a deadlock after doing a backtrace. This is shown below.

Various steps were carried out to find the root cause of this issue:

* The gem5 KVM setup guide (https://www.gem5.org/documentation/general_docs/using_kvm/) was followed exactly on both systems and nothing indicated KVM was not setup correctly on each machine. There was still no change in the observed behavior.
* The same binary was compiled in the Working system and copied to to the Newer system. No change.
* The i7-8700 system was updagraded to Ubuntu 22.04 and the deadlock occurred.
* The i7-13700 system was downgraded to Linux 5.19 and deadlock didn't occur.
* QEMU and virtualbox show KVM running fine on the i7-13700 system despire this problem with gem5.
* An i7-10510u system was found to run the same simulation without issue.
* To 100% rule out any environmental factors (OS, libraries, applications, anything) a Live USB stick was created to run the entire OS, gem5 directory, build, etc. on both machines. Still the i7-13700 deadlocked and the i7-8700 system didn't.

It could be the bios settings but the user couldn't think of anything there which was incorrect and didn't bother to change anything to find out.

**Affects version**

The was working on [b271367a357cbb390084884e809235d6d62b9cf5](https://github.com/gem5/gem5/commit/b271367a357cbb390084884e809235d6d62b9cf5), which is v22.1.0.0.

This has yet to be observed checked on v23.0.

**To Reproduce**

This has yet to be replicated but based on our assumptions:

1. Follow the setting up KVM guide on an i7-13700 machine: https://www.gem5.org/documentation/general_docs/using_kvm/
2. Clone gem5 and checkout at v22.1.0.0.
3. Compile X86/gem5.opt: `scons build/X86/gem5.opt -j <threads>`.
4. Run `./build/X86/gem5.opt gem5/configs/example/gem5_library/x86-ubuntu-run-with-kvm.py`.

The simulation should reach a point and hang indefinitely.

**Backtrace**

The deadlock was determined by running a backtrace. The following was observed:

```sh
#4  ___pthread_cond_wait (cond=0x55555951f740, mutex=0x55555951f718) at ./nptl/pthread_cond_wait.c:627
#5  0x000055555648f655 in gem5::Barrier::wait (this=0x55555951f718) at build/X86/base/barrier.hh:78
#6  0x000055555648f753 in gem5::BaseGlobalEvent::BarrierEvent::globalBarrier (this=0x55555ee59280) at build/X86/sim/global_event.hh:104
#7  0x000055555648f3f8 in gem5::GlobalSyncEvent::BarrierEvent::process (this=0x55555ee59280) at build/X86/sim/global_event.cc:147
#8  0x0000555556483a6f in gem5::EventQueue::serviceOne (this=0x555558bcaa20) at build/X86/sim/eventq.cc:223
#9  0x00005555564b2c7b in gem5::doSimLoop (eventq=0x555558bcaa20) at build/X86/sim/simulate.cc:340
#10 0x00005555564b26f4 in gem5::simulate (num_cycles=18446744073709551615) at build/X86/sim/simulate.cc:227
```
**Host Operating System**

Ubuntu/Linux. Various versions tried though the OS, ultimately, don't seem to be a factor as mentioned above.

**Host ISA**

X86

**Compiler used**

gcc 11.4 in all cases.

**Additional information**

Python 3.10 used in all cases.

atrib, 2023-09-04:
Hi, I'm the user who reported this issue. 

Just chiming in with a little correction:
- The i7-8700 was upgraded to Ubuntu 22.04 and the deadlock **did not occur**
- The i7-13700 was downgraded to kernels 5.4 and 5.19, and the deadlock **occurred**.

I checked the BIOS settings, to verify that VT-x is enabled, as is "VT for direct I/O". Intel TXT is disabled. No other virtualization option seemed relevant to the issue. SecureBoot is disabled (related to a different project).

I also tried the `develop` branch (can't remember the commit ID, but it was around Aug 27th).

To add to the steps to reproduce:
1. In the KVM guide, `apt` replaces the package `qemu-kvm` with `qemu-system-x86`
2. Same
3. Install gem5 dependencies from https://www.gem5.org/documentation/general_docs/building.
4. Same.

Please feel free to reach out if any further info is needed.

ivanaamit, 2024-01-08:
We are closing this issue due to a lack of activity.

closed 2024-01-08 by ivanaamit