gem5/gem5 #266: Website is not displaying correctly.

cnjsdfcy, 2023-09-05:
Hi there,

I'm not sure if it's just me or someone else also encountering the website displaying problem, please help take a look since it's preventing smooth reading of the documents. Thanks~

[image]

Harshil2107, 2023-09-07:
Hi,
Can you send the link to the page you are viewing? I checked the documentation page of the gem5 website and the footer is being rendered differently for me. I have attached a screenshot of how it looks on my system.

[image]

cnjsdfcy, 2023-09-12:
Hi @Harshil2107 ,

The link is https://www.gem5.org , and I tried both Chrome and Edge, both had the same rendering issue.

Thanks.

Harshil2107, 2023-09-20:
I looked on my end, The website looks the same as the screenshot I sent above. I tried different aspect ratios and window sizes, but I couldn't reproduce the rendering issue you were having. 

Are there any other details that you can give to help me reproduce the issue?

Thanks.

cnjsdfcy, 2023-09-26:
Hi @Harshil2107 ,

After fixed the DNS of company, the web is rendering OK.

Thanks and closing the ticket.

closed 2023-09-26 by cnjsdfcy