gem5/gem5 #289: Non-deterministic simulation results

ivanaamit, 2023-09-08:


BobbyRBruce, 2023-09-11:
I created #300 to add some tests to find this. In doing so I observed the following (note: I use the `x86-ubuntu-run-to-tick.py` script which is in the #300):

```sh
./build/ALL/gem5.opt --stats-file=json://stats-baseline-1.json configs/example/gem5_library/x86-ubuntu-run-to-tick.py --ticks 1000000
./build/ALL/gem5.opt --stats-file=json://stats-baseline-2.json configs/example/gem5_library/x86-ubuntu-run-to-tick.py --ticks 1000000
diff <(jq 'del(.creation_time)' stats-baseline-1.json | jq --sort-keys .) <(jq 'del(.creation_time)' stats-baseline-2.json | jq --sort-keys .)
```

The following is output (top 20 lines):

```sh
14247c14247
<             "underflow": 2.317153539e-314,
---
>             "underflow": 2.298907924e-314,
14281c14281
<             "underflow": 2.3171548436e-314,
---
>             "underflow": 2.2989092285e-314,
14335c14335
<             "underflow": 2.3254696917e-314,
---
>             "underflow": 2.3072240767e-314,
14358c14358
<             "overflow": 2.3285561396e-314,
---
>             "overflow": 2.3103105245e-314,
14362c14362
<             "underflow": 2.3171540847e-314,
---
>             "underflow": 2.2989084696e-314,
```

Few comments:

1. Why is "overflow" and "underflow" being set to these _insane_ values?
2. Even taking this into account, with floating-point weirdness, these values should be deterministic and not change between runs.

I think we have some undefined behavior here.

**To clarify:** This is the same compiler and same build of gem5.

powerjg, 2023-09-11:
I think this might be a json printing problem, and I think it's different than what Ivana initially posted. I wonder if you didn't use json if this problem would go away.

BobbyRBruce, 2023-09-14:
> I think this might be a json printing problem, and I think it's different than what Ivana initially posted. I wonder if you didn't use json if this problem would go away.

You're right. I don't know really why... I still think it should be deterministic.

I've updated #300 to use the txt file instead. It could be incorporated to see how our gem5 compilations differ.

closed 2023-10-31 by github-actions[bot]