gem5/gem5 #292: Add clang 15 and clang 16 to the compiler tests

mkjost0, 2023-09-08:
Currently, our compiler tests don't check versions 15 and 16 of clang, so we need to create new Docker images for this and make sure they're added to our tests.  In addition, once these are added, we can remove older versions of clang as to not overrun our testing infrastructure.

MeatBoy106, 2023-10-13:
Clang-17 is also out. Should it be added as well?

BobbyRBruce, 2023-10-16:
> Clang-17 is also out. Should it be added as well?

I'm actually about to close this issue because clang 15 and 16 have been incorporated (just not really run yet... i'll hold off just to make sure they do.).

I'm personally ok not trying to hard to keep up with clang release given it's rapid pace of development. More-so though our compiler tests are still failing and don't want to add more tests until we fix the tests we're running.

We will add it eventually though.

closed 2023-10-18 by Harshil2107