gem5/gem5 #311: GPU GL2 atomics failing

abmerop, 2023-09-13:
**Describe the bug**
I am creating this issue because I want to make sure it is tracked and fixed before the next stable release.  Desc: GPU local atomics (in the L2 caches of a GPU) are currently failing two of the now deprecated weekly tests.

**Affects version**
1bebf6a3cc (upstream/develop, origin/develop, github-develop, develop) sim-se: Use tgt_stat64 instead of tgt_stat in newfstatatFunc (#283)

**gem5 Modifications**
No modifications

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:
1. Compile gem5 with command:   scons --linker=mold  -j126 build/VEGA_X86/gem5.opt -k
2. build/VEGA_X86/gem5.opt configs/example/gpufs/vega10_kvm.py --disk-image <path>/ubuntu20-rocm54 --kernel <path>/vmlinux-5.8.0-ubuntu-config --gpu-mmio-trace <path>/gem5-resources/src/gpu-fs/vega_mmio.log  --app  <path>/gem5-resources/src/gpu/pannotia/bc/bin/bc_hip --opts="/home/gem5/1k_128k.gr"

The disk image does not matter. It happens in any ROCm version due to bug being in the TCC / SLICC code. Alternate SE mode:

1a. docker run -u $UID:$GID -w <gem5_cwd> gcr.io/gem5-test/gcn-gpu:latest scons -j126 build/VEGA_X86/gem5.opt -k
2a. docker run -u $UID:$GID -w <gem5_cwd> gcr.io/gem5-test/gcn-gpu:latest build/VEGA_X86/gem5.opt configs/example/apu_se.py -n8  -c  <path>/gem5-resources/src/gpu/pannotia/bc/bin/bc.gem5  --options="1k_128k.gr"  --dgpu --gfx-version=gfx900

**Terminal Output**
simerr output: 
build/VEGA_X86/mem/ruby/protocol/TCC_Transitions.cc:768: panic: Invalid transition                                                                                                                                                                                                  system.ruby.tcc_cntrl0 time: 36221253233 addr: 0xc020c0 event: WrVicBlk state: M

**Expected behavior**
Expect application to run.  It works on stable v23.

**Host Operating System**
Two: Ubuntu 20.04 and Ubuntu 22.04

**Host ISA**
amd64

**Compiler used**
gcc 9.4.0

**Additional information**
Most likely caused by PR #101

mattsinc, 2023-09-13:
@DanKouch just making sure you saw this

mattsinc, 2023-09-20:
@DanKouch reminder to please update this chain with what we discussed

abmerop, 2023-09-20:
Not sure if y'all have started to work on this, but this minimal change fixes the issue mentioned here, but there are still other issues.  Namely, the protocol tester with TCC (GL2) atomics.  You can try these again by reverting #133. I'm also seeing one failure in FS mode with an invalid transition: AtomicDone in M state.

`--- a/src/mem/ruby/protocol/GPU_VIPER-TCC.sm
+++ b/src/mem/ruby/protocol/GPU_VIPER-TCC.sm
@@ -924,6 +924,14 @@ machine(MachineType:TCC, "TCC Cache")
     p_popRequestQueue;
   }

+  transition(M, WrVicBlk) {TagArrayRead, DataArrayWrite} {
+    p_profileHit;
+    ut_updateTag;
+    wdb_writeDirtyBytes;
+    wt_writeThrough;
+    p_popRequestQueue;
+  }
+
   transition({V, M}, WrVicBlkBack, M) {TagArrayRead, TagArrayWrite, DataArrayWrite} {
     p_profileHit;
     ut_updateTag;
@@ -1061,7 +1069,7 @@ machine(MachineType:TCC, "TCC Cache")

   //M,W should not see WBAck as the cache is in WB mode
   //WBAcks do not need to check tags
-  transition({I, V, IV, A}, WBAck) {
+  transition({I, V, IV, A, M}, WBAck) {
     w_sendResponseWBAck;
     pr_popResponseQueue;
   }`

This is just one way to fix this.  It assumes that if there is a dirty copy in the L2, an L1 write through will modify the L2's local copy *and* pass the data on to the directory.  Alternately you could invalidate the L2 first which would require a lot more effort.  There is of course no public documentation on proprietary cache protocols so your guess is as good as mine. I'll leave if to you to choose unless you would really like to discuss it.

closed 2023-11-27 by Harshil2107