gem5/gem5 #312: Specifying mold linker does not actually use mold linker

abmerop, 2023-09-13:
**Describe the bug**
Specifying mold linker does not actually use mold linker

**Affects version**
develop, starting after #244 

**gem5 Modifications**
No modifications

**To Reproduce**
scons --linker=mold  -j24 build/VEGA_X86/gem5.opt -k  --mold-path=/usr/local/bin/mold

See notes about Ubuntu 20 below.

If writing code, or a terminal command, use code blocks. Either an inline code block, \`scons build/ALL/gem5.opt\` (enclosed in two \`) or a multi-line codeblock:

**Expected behavior**
This:

[image]

Not this:

[image]

**Host Operating System**
Ubuntu 20.04

**Host ISA**
amd64

**Compiler used**
gcc 9.4.0

**Additional information**
Ubuntu 20.04 (in contrast to 22.04) does not have the mold linker as part of the apt / snap.  Therefore, most people built from source.  Like most linux configure && make && make install, this goes in /usr/local.  This worked just fine util #244 . Specifying the direct path should just work in my opinion and requesting users to 'just upgrade to 22.04' is not a reasonable option.

abmerop, 2023-09-13:
Relevant gem5 blog post to demonstrate the importance: https://www.gem5.org/project/2023/02/16/benchmarking-linkers.html

powerjg, 2023-09-13:
@hnpl, should we revert #244?

BobbyRBruce, 2023-09-13:
I created a revert: https://github.com/gem5/gem5/pull/313

BobbyRBruce, 2023-09-14:
#313 effectively fixes this issue, so I'm closing it. I'll leave it to @hnpl to come up with a better solution.

PR #313 (2023-09-14): scons: Revert "Add an option specifying the path to mold linker binary"
Reverts gem5/gem5#244

Fixes #312
closed 2023-09-14 by BobbyRBruce