gem5/gem5 #330: Segmentation Fault in GPUFS in modprobe Section of Runscript

Harshil2107, 2023-09-19:
This issue is made from a Jira post:https://gem5.atlassian.net/browse/GEM5-1334?atlOrigin=eyJpIjoiYjQwNjAwZDExOWJlNGM3Zjg4NjBmZDQ4ZTNkZTcyYzAiLCJwIjoiaiJ9

Below is the issue as described on Jira:

I am running the FS GPU as described in this tutorial/article:
https://www.gem5.org/2023/02/13/moving-to-full-system-gpu.html

I am using an unmodified version of gem5 v23. But I also tested with same results with gem5 v22.0

The cmd run is
```
./build/VEGA_X86/gem5.fast configs/example/gpufs/hip_rodinia.py --kernel gpufs_disk_image/vmlinux-5.4.0-105-generic --gpu-mmio-trace gpufs_disk_image/vega_mmio.log --disk-image gpufs_disk_image/x86-gpu-fs-20220512.img -a nn
```
where gpufs_disk_image has downloads for the kernel, disk image, and mmio.log from gem5-resources. Links below in “environment” section. I also tried specifying the gpu-rom as the vega10.rom given in resources but that did not solve/change the error.

 

At the end of the day, none of the applications (even “known working” nn) run correctly. All of them seg fault when the binary is executed. There is an earlier seg fault in the modprobe which I believe causes the error later. The later segmentation fault occurs after executing the benchmark. Within the guest…

```
make test
```

> executing: ../../test/nn/run0.cmd...    [   48.587003] nn[624]: segfault at 68 ip 00007f6464ce1baf sp 00007ffedff1ae90 error 4 in libamdhip64.so.4.2.40200[7f6464c3c000+256000]
[   48.591003] Code: 00 48 8b 88 38 10 00 00 4d 85 c0 74 18 be 66 00 00 00 bf 03 00 00 00 41 ff d0 48 8b 45 88 c7 40 08 01 00 00 00 e8 c1 b2 fa ff <48> 8b 40 68 48 8b 40 18 48 8b 00 48 85 c0 49 89 c7 0f 84 0a 01 00
Segmentation fault
  FAILED!
  
I believe the issue is with the command, as written in the runscript in hip_rodinia/samples.py . From within the guest the following was run.

```
modprobe -v amdgpu ip_block_mask=0xff ppfeaturemask=0 dpm=0 audio=0
```
which returns the following (entire system trace/log attached as file to this issue) from within the FS simulation: 

> [   70.236353] amdkcl: loading out-of-tree module taints kernel.
[   70.240354] amdkcl: module verification failed: signature and/or required key missing - tainting kernel
[   70.280356] AMD-Vi: AMD IOMMUv2 driver by Joerg Roedel [jroedel@suse.de](mailto:jroedel@suse.de)
[   70.284356] AMD-Vi: AMD IOMMUv2 functionality not available on this system
[   70.456367] [drm] amdgpu kernel modesetting enabled.
[   70.460367] [drm] amdgpu version: 5.9.25
[   70.464368] [drm] OS DRM version: 5.4.0
[   70.468368] amdgpu: CRAT table not found
[   70.472368] amdgpu: DSDT table not found for OEM information
[   70.476368] amdgpu: Virtual CRAT table created for CPU
[   70.480369] amdgpu: Topology: Add CPU node
[   70.508370] amdgpu: PeerDirect support was initialized successfully
[   70.512371] amdgpu 0000:00:08.0: remove_conflicting_pci_framebuffers: bar 0: 0x400000000 -> 0x7ffffffff
[   70.516371] amdgpu 0000:00:08.0: remove_conflicting_pci_framebuffers: bar 2: 0x100000000 -> 0x1001fffff
[   70.520371] amdgpu 0000:00:08.0: remove_conflicting_pci_framebuffers: bar 5: 0xc0000000 -> 0xc007ffff
[   70.528372] amdgpu 0000:00:08.0: vgaarb: deactivate vga console
[   70.532372] amdgpu 0000:00:08.0: can't find IRQ for PCI INT B; probably buggy MP table
[   70.536372] [drm] initializing kernel modesetting (VEGA10 0x1002:0x6863 0x0000:0x0000 0x00).
[   70.540372] amdgpu 0000:00:08.0: amdgpu: Trusted Memory Zone (TMZ) feature not supported
[   70.544373] [drm] register mmio base: 0xC0000000
[   70.544373] [drm] register mmio size: 524288
[   70.548373] [drm] PCI I/O BAR is not found.
[   70.552373] [drm] PCIE atomic ops is not supported
[   70.556373] [drm] add ip block number 0 <soc15_common>
[   70.556373] [drm] add ip block number 1 <gmc_v9_0>
[   70.560374] [drm] add ip block number 2 <vega10_ih>
[   70.564374] [drm] add ip block number 3 <psp>
[   70.568374] [drm] add ip block number 4 <gfx_v9_0>
[   70.568374] [drm] add ip block number 5 <sdma_v4_0>
[   70.572374] [drm] add ip block number 6 <powerplay>
[   70.576375] [drm] add ip block number 7 <dm>
[   70.576375] [drm] add ip block number 8 <uvd_v7_0>
[   70.580375] [drm] add ip block number 9 <vce_v4_0>
[   70.592376] amdgpu 0000:00:08.0: No more image in the PCI ROM
[   70.592376] amdgpu 0000:00:08.0: amdgpu: Fetched VBIOS from ROM BAR
[... 140 lines cut ...]
cat /sys/module/amdgpu/parameters/ip_block_mask
cat /sys/module/amdgpu/parameters/ppfeaturemask
cat /sys/module/amdgpu/parameters/dpm
cat /sys/module/amdgpu/parameters/audio
```
> 255
0
0
0

So it seems it’s unable to find the shared object file libamdhip64.so.4.2.40200. I imagine that is expected to be loaded during the modprobe.

Thank you for the help. I hope the logs and information provided are sufficient. I put a somewhat related issue/Jira ticket.

## Environment

disk image: http://dist.gem5.org/dist/v22-1/images/x86/ubuntu-18-04/x86-gpu-fs-20220512.img.gz

kernel: http://dist.gem5.org/dist/v22-1/kernels/x86/static/vmlinux-5.4.0-105-generic

mmio.log: https://github.com/gem5/gem5-resources/blob/stable/src/gpu-fs/vega_mmio.log  

## Files attached
[guest_log.txt](https://github.com/gem5/gem5/files/12666791/guest_log.txt)
[host_log.txt](https://github.com/gem5/gem5/files/12666792/host_log.txt)

Harshil2107, 2023-09-19:
@abmerop can you take a look at this, Matt Sinclair asked me to assign this to you on Jira.

abmerop, 2023-09-19:
It doesn't look like you did anything wrong.  This seems to be the issue real issue:

> [ 71.260417] invalid opcode: 0000 # 1 SMP NOPTI

The guest Linux kernel (5.4.0-105-generic) seems to be crashing when running in KVM on your machine.  In particular, on a clflush instruction with a lock prefix (I think) in Linux' memory management code looking at the function name / opcode dump.  What hardware does your host machine have?  Is clflush listed in `cat /proc/cpuinfo`?

You can try on the `develop` branch as well. 

The driver is not fully loaded, but might be trying to reset (which gem5 doesn't support).  The application will crash on any API call to the driver if it's not loaded properly.

abmerop, 2023-09-20:
Another possibility to rule out KVM:

Modify hip_rodinia.py and change line 132: 
     args.cpu_type = "X86KvmCPU" 

to
     args.cpu_type = "AtomicSimpleCPU"

This took 8 hours last time I tried it, but it also compiles `nn` which takes most of the time and really we're just looking to see if the driver loads with this cpu type.  That should take around 2 hours with that CPU in this config.  It's something that could possibly run in the background / overnight.

cgreen18, 2023-09-22:
Ah, you're right the issue is with KVM and the hardware. I am the one who posted the original Jira ticket (copied here).

I eventually found a machine where the code executed without fault and the machine had a kernel version "Linux 6.2.0-33-generic x86_64" while one's that failed had versions < 5 (3.10, 4.15).
All machines contained "clflush" in /proc/cpuinfo but the one that worked also had "clflushopt"
`cat /proc/cpuinfo` returns `flags		: ... clflush ... clflushopt ...` means the KVM works
`cat /proc/cpuinfo` returns `flags		: ... clflush ... ` did not work
So possibly the required opcode is only valid for machines with "clflushopt"

I also ran with AtomicCPU and it worked as well. Thank you for the debugging help.

abmerop, 2023-09-22:
That's great, I am glad it worked!  I think there may still be some bug here though, which is the KVM setting CPUID flags which the host doesn't support.  I am not sure the best way to fix that off the top of my head, but I recent pushed a change which can set the CPUID flags from python, which could be useful there.

cgreen18, 2023-09-27:
Okay that's great. For now, I'll make checkpoints and use the hardware that works. Later, I'll check out the python support for CPUIDs. Thank you very much, I think it's resolved but I'm not sure if I can flag this issue as resolved as I did not post (was copied over)

powerjg, 2023-09-27:
Thanks, @cgreen18 I'll close the issue.

closed 2023-09-27 by powerjg