gem5/gem5 #349: gem5 RISCV Full System Simulation w/ Ubuntu 22.04 Failed to Execute STREAM+OpenMP+clang-14/gcc-12 properly

hnpl, 2023-09-21:
**Describe the bug**
gem5 RISCV Full System Simulation w/ Ubuntu 22.04 Failed to Execute STREAM+OpenMP+clang14 properly using STREAM with 1 thread and gem5 in atomic memory mode.

**Affects version**
develop for v23.1
https://github.com/gem5/gem5/tree/f5a255c68d1ce0318da0d46f744ee7513e0acda9

**gem5 Modifications**
No

**To Reproduce**
Step 1. Create the Ubuntu 22.04 disk image as showed here, https://github.com/hnpl/hpc-disk-image#4-building-the-rv64gc-disk-image.
Step 2. In the guest system, install gcc-12 or clang-14 and the related tools.
Step 3. Recompiling the STREAM benchmark in the guest `/home/ubuntu/simple-vectorizable-microbenchmarks/stream` with the following command,
```sh
(sudo?) make -f makefiles/Makefile-hw M5_BUILD_PATH=/home/ubuntu/gem5/util/m5/build/riscv/ M5OPS_HEADER_PATH=/home/ubuntu/gem5/include/
```
Optionally, if you're using clang, you'd need to modify the file `/home/ubuntu/simple-vectorizable-microbenchmarks/stream/makefiles/Makefile-hw`.
Step 4. Run the following command,
```sh
build/RISCV/gem5.opt -re --outdir=atomic-500000 riscv-fs-test.py --command "/home/ubuntu/simple-vectorizable-microbenchmarks/stream/stream.rv64gc.m5 500000"
```

gem5 Python configuration,

```Python
import m5

from gem5.components.boards.riscv_board import RiscvBoard
from gem5.components.memory import SingleChannelDDR3_1600
from gem5.components.processors.simple_processor import SimpleProcessor
from gem5.components.cachehierarchies.classic.private_l1_private_l2_cache_hierarchy import (
    PrivateL1PrivateL2CacheHierarchy,
)
from gem5.components.processors.cpu_types import CPUTypes
from gem5.isas import ISA
from gem5.utils.requires import requires
from gem5.utils.override import overrides
from gem5.resources.resource import Resource, CustomResource, DiskImageResource, FileResource
from gem5.simulate.simulator import Simulator

# Run a check to ensure the right version of gem5 is being used.
requires(isa_required=ISA.RISCV)

# Setup the cache hierarchy.
# For classic, PrivateL1PrivateL2 and NoCache have been tested.
# For Ruby, MESI_Two_Level and MI_example have been tested.
cache_hierarchy = PrivateL1PrivateL2CacheHierarchy(
    l1d_size="32KiB", l1i_size="32KiB", l2_size="512KiB"
)

# Setup the system memory.
memory = SingleChannelDDR3_1600()

# Setup a single core Processor.
processor = SimpleProcessor(
    cpu_type=CPUTypes.ATOMIC, isa=ISA.RISCV, num_cores=1
)

class ModifiedRiscvBoard(RiscvBoard):
    @overrides(RiscvBoard)
    def get_default_kernel_args(self):
        return [
            "earlyprintk=ttyS0",
            "console=ttyS0",
            "lpj=7999923",
            "root=/dev/vda1",
            "init=/root/gem5-init.sh",
            "rw",
        ]

# Setup the board.
board = ModifiedRiscvBoard(
    clk_freq="1GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=cache_hierarchy,
)

import argparse
parser = argparse.ArgumentParser()
parser.add_argument("--command", type=str)
args = parser.parse_args()

command = args.command

# Set the Full System workload.
board.set_kernel_disk_workload(
    kernel=FileResource("/scr/hn/bbl"),
    disk_image=DiskImageResource("/scr/hn/rv64gc-hpc-2204.img"),
    readfile_contents=f"{command}"
)

simulator = Simulator(board=board)
#m5.debug.flags["ExecAll"].enable()
#m5.debug.flags["LLSC"].enable()
print("Beginning simulation!")
# Note: This simulation will never stop. You can access the terminal upon boot
# using m5term (`./util/term`): `./m5term localhost <port>`. Note the `<port>`
# value is obtained from the gem5 terminal stdout. Look out for
# "system.platform.terminal: Listening for connections on port <port>".
simulator.run()
print("exit 1")
m5.debug.flags["ExecAll"].enable()
m5.debug.flags["LLSC"].enable()
#m5.debug.flags["RiscvMiscFS"].enable()
simulator.run()
print("exit 2")
#simulator.run()
#print("exit 3")

```

**Terminal Output**
N/A

**Expected behavior**
[Expection] The `board.platform.terminal` file should show,

```
Number of threads: 1
...
array_a error_count: 0, max_error: 0.000000
array_b error_count: 0, max_error: 0.000000
array_c error_count: 0, max_error: 0.000000

```

[Reality] The `board.platform.terminal` file shows non-zero error_counts, e.g.,
```
Number of threads: 1
...
array_a error_count: 352070, max_error: 15.000000
array_b error_count: 352070, max_error: 3.000000
array_c error_count: 352070, max_error: 4.000000

```

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**
N/A

hnpl, 2023-09-21:
More info,

- STREAM consists of the following operations,
```
a = vector<double>(SIZE, INITIAL_VALUE_A)
b = vector<double>(SIZE, INITIAL_VALUE_B)
c = vector<double>(SIZE, INITIAL_VALUE_C)
etc.
```

- The bug happened when initializing vector `a`, where software copies a floating point register to memory. This bug happened as follows,
```
store fa5, mem[EA1]  // the content of fa5 is 246.0
<context_switch_started>
load fa5, mem[EA2] // the content of fa5 is 0.0, this operation is not supposed to happen
<context_switch_end>
store fa5, mem[EA3] // the content of fa5 is 0.0, but fa5 should be 246.0 instead
```

aarmejach, 2023-09-27:
We are not experiencing these issues, but recall similar problems a few months ago.

IIRC these were solved after we upgraded system software by moving to an Ubuntu 23.04 disk image. Using gcc13.1 and clang15+ on Ubuntu 23.04 has been working quite well for us in the past month. 

With Ubuntu 22.04 we experienced problems with OpenMP, so I'd suggest moving to Ubuntu 23.04 and try if the same behaviour appears.

hnpl, 2023-09-27:
@aarmejach Thanks for the information, I'll try the newer compilers. Can you give more details on which version of the Linux kernel are you using?

hnpl, 2023-09-27:
I just ran the STREAM benchmark again with Linux kernel 6.5 + Ubuntu 23.04 + gcc13. The above problem persists!

The symptom of this problem is somewhere in the Linux kernel, fstate_restore() was called before fstate_save() was called.

hnpl, 2023-09-28:
I justed tested STREAM with the bbl+vmlinux kernel from gem5 resources and the Ubuntu 23.04 userland. The setup worked correctly.

Thanks @aarmejach for pointing this out. I'll close this issue.

aarmejach, 2023-09-28:
Hi @hnpl, that's great. 
Could you share how you managed to boot with OpenSBI 1.3 + Linux kernel 6.5 in gem5?
We are interested in using v6.5 since it's the first version with RVV kernel support. I tryied booting it but got stuck during boot.

hnpl, 2023-09-28:
I booted the Linux kernel with OpenSBI FW_JUMP using the atomic CPU. I noticed that booting with more cores will cause gem5 to run significantly slower compared to running with fewer cores. gem5 with 1 core managed to boot the Linux kernel within ~15 mins, while gem5 with 16 cores managed to boot within ~2 hours. Interestingly, with 64 cores, gem5 never managed to enter the kernel.

I can make a PR if you're interested in using OpenSBI+Linux 6.5.

hnpl, 2023-09-29:
The problem appears again with multiple cores. I'm going to reopen this issue.

hnpl, 2023-09-29:
#370 should fix this

hnpl, 2023-09-29:
This probably is a problem in Vector Instruction as well given how the Linux kernel handles Vector Registers save/restore

https://elixir.bootlin.com/linux/latest/source/arch/riscv/include/asm/vector.h#L165

aarmejach, 2023-09-29:
> I booted the Linux kernel with OpenSBI FW_JUMP using the atomic CPU. I noticed that booting with more cores will cause gem5 to run significantly slower compared to running with fewer cores. gem5 [...]

A PR would be great, it is the only way to get RVV support in FS. With the current 5.10 kernel RVV support is actually disabled (`STATUS` register bits for RVV are set to off at boot, there is no V register state handling on context switch, etc.).

Yes, for vector we will have to put the STATUS bits to dirty for V register, which will trigger proper save/restore on the kernel side. But for that we need kernel v6.5+.

BobbyRBruce, 2023-09-29:
Absolutely incredible work here.  Great job @hnpl!

This issue appears fixed by #370. Closing Issue.

closed 2023-09-29 by BobbyRBruce