gem5/gem5 #368: Add a finite snoop filter model in ruby CHI implementation

JoshRandall, 2023-09-28:
This involves adding directory eviction and back-invalidation support to the protocol as well as the finite snoop filter implementation.

powerjg, 2023-09-28:
Letting @giactra and @tiagormk know that you're working on this.

ivanaamit, 2024-01-16:
Hi @JoshRandall, I'm checking to see if you're currently working on this. Thank you.

JoshRandall, 2024-01-17:
Hi @ivanaamit, I am still working on this.

ivanaamit, 2024-04-12:
Hi @JoshRandall, what is the current status of this?

We are creating the staging branch for the upcoming release on May 20th. All the changes intended for this release need to be merged before that date. Are you actively working on this and planning to finish it in the next month? Thank you.

JoshRandall, 2024-04-18:
Still working on this, I should have it out soon.

ivanaamit, 2024-05-13:
Hi @JoshRandall, just a reminder that we are creating a staging branch in a week (May 20th). You should create a PR in the next day or two if you want this change to be incorporated into gem5 v24.0.

JoshRandall, 2024-05-13:
Hi @ivanaamit, thanks for letting me know, this won't be ready for a PR in that time frame.

erin-le, 2025-03-07:
@JoshRandall: Hello, thank you for opening this issue. Are you still working on this feature?

erin-le, 2025-04-24:
@JoshRandall: This PR is being closed for inactivity. If you plan to continue working on this feature, please feel free to reopen. Thank you!