gem5/gem5 #373: Add PerThreadUnifiedRenameMap Type to O3 CPU

mkjost0, 2023-09-29:
Currently, to allow clang-15 and clang-16 to compile in #365 , we added MaxThreads to the UnifiedRenameMap in the function setRenameMap.  However, for a cleaner solution to this, we can introduce a PerThreadUnifiedRenameMap type (or something along those lines) that is a std::array<UnifiedRenameMap, MaxThreads>. That would make it behave as a normal C++ container but with the performance of a plain array.

Arnab035, 2024-01-31:
Hello,

I just completed a project wherein I interacted with the memory subsystem in gem5 and I would like to contribute to the official repository. May I assign myself this task?

powerjg, 2024-01-31:
Sure! Please reference this issue when creating your PR. No need to "assign" the task.

Arnab035, 2024-02-12:
Hello,

May I close this issue with a closing message where I link the PR?

BobbyRBruce, 2024-02-12:
> Hello,
> 
> May I close this issue with a closing message where I link the PR?

Of course!

Arnab035, 2024-02-12:
Fixed by #842

Arnab035, 2024-02-12:
> > Hello,
> > May I close this issue with a closing message where I link the PR?
> 
> Of course!

Oh, it seems I do not have permissions to close. Would you please help me in closing this issue?

BobbyRBruce, 2024-02-12:
> > > Hello,
> > > May I close this issue with a closing message where I link the PR?
> > 
> > 
> > Of course!
> 
> Oh, it seems I do not have permissions to close. Would you please help me in closing this issue?

No problem. I guess you have to have created the issue to close it if you're not a maintainer. It's now closed.

closed 2024-02-12 by BobbyRBruce