gem5/gem5 #374: Crash when pinging other simulation in Arm dist_bigLITTLE with SMMU

sammy17, 2023-09-29:
**Describe the bug**
I was able to get arm dist_bigLITTLE setup running with SMMU as discussed in https://github.com/orgs/gem5/discussions/239
The specific changes I made are mentioned in the gem5 modifications section. 
My goal is to measure network throughput of the ethernet link between two simulations when the SMMU is present. 
Therefore, I am using timing CPU models with two Minor CPUs in the dist setup with caches enabled. 

Linux booted without any error and SMMU was recognized by Linux. 
After I setup the ethernet connections of each simulation using ifconfig, I tried to ping one simulation from the other, which caused a crash with the below error:
```
11956396690000: system.membus.snoop_filter: eraseIfNullEntry:   Removed SF entry.
11956396702000: system.realview.smmu.device_interfaces: [t] req from system.ethernet.dma addr=0xfffff000 size=0x10
11956396704000: system.realview.smmu.device_interfaces.port: micro TLB hit vaddr=0xfffff000 amask=0xfffffffffffff000 sid=0 ssid=0 paddr=0x81d90000
11956396705000: system.realview.smmu: [t] requestor req  addr=0x81d90000 size=0x10
11956396705000: system.membus.snoop_filter: lookupRequest: src system.membus.cpu_side_port[4] packet WriteReq [81d90000:81d9000f]
11956396705000: system.membus.snoop_filter: lookupRequest:   SF value 0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010
11956396705000: system.bigCluster.toL2Bus.snoop_filter: lookupSnoop: packet WriteReq [81d90000:81d9000f] ES
gem5.opt: src/mem/snoop_filter.cc:225: std::pair<std::vector<gem5::QueuedResponsePort*>, gem5::Cycles> gem5::SnoopFilter::lookupSnoop(const gem5::Packet*): Assertion `cpkt->isWriteback() || cpkt->req->isUncacheable() || (cpkt->isInvalidate() == cpkt->needsWritable()) || cpkt->req->isCacheMaintenance()' failed.
Program aborted at tick 11956396705000
--- BEGIN LIBC BACKTRACE ---
```
It should be noted that without the SMMU, pinging and other network communication between two simulations work fine. 
Please refer to gem5 modifications section to see the changes made to enable SMMU in dist_bigLITTLE setup. 

**Affects version**
v23.0.1.0

**gem5 Modifications**
https://github.com/gem5/gem5/blob/48a40cf2f5182a82de360b7efa497d82e06b1631/configs/example/arm/devices.py#L461
attach_pci function was changed to add *args, **kwargs and pass **kwargs as below:
```
def attach_pci(self, dev, *args, **kwargs):
        self.realview.attachPciDevice(dev, self.iobus, **kwargs)
```
https://github.com/gem5/gem5/blob/48a40cf2f5182a82de360b7efa497d82e06b1631/configs/example/arm/dist_bigLITTLE.py#L138
addEthernet function was changed to instantiate SMMU and connect the Ethernet device to SMMU as below:
```
def addEthernet(system, options):
        # Create an SMMU but don't connect it to any devices.
        system.realview.attachSmmu([], system.membus)
        system.realview.smmu.irq_interface_enable = True
        # create NIC
        dev = IGbE_e1000()
        system.attach_pci(dev, dma_ports=[])
        system.ethernet = dev
        # Connect the device's DMA ports to the SMMU
        system.realview.smmu.connect(dev)
```
A manually edited device tree was used to include the SMMU and map the PCI ethernet device to the SMMU/IOMMU. 
Following are the relevant device tree nodes of SMMU and PCI host. 
```
	iommu: smmuv3@2b400000 {
		compatible = "arm,smmu-v3";
		reg = <0x00 0x2b400000 0x00 0x20000>;
		#iommu-cells = <0x01>;
		phandle = <0x11>;
		dma-coherent;
	};

	pci {
		compatible = "pci-host-ecam-generic";
		device_type = "pci";
		#address-cells = <0x03>;
		#size-cells = <0x02>;
		#interrupt-cells = <0x01>;
		reg = <0x00 0x40000000 0x00 0x10000000>;
		ranges = <0x1000000 0x00 0x00 0x00 0x50000000 0x00 0x10000 0x2000000 0x00 0x00 0x40 0x00 0x00 0x40000000>;
		interrupt-map = <0x00 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x44 0x01 0x800 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x45 0x01 0x1000 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x46 0x01 0x1800 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x47 0x01>;
		interrupt-map-mask = <0x1800 0x00 0x00 0x00>;
		iommu-map = <0x10 &iommu 0x0 0x1>;
		dma-coherent;
	};
```
Note that in both SMMU and pci host, dma-coherent feature is enabled. 
I kept dma-coherent for both these devices as I got to know that IO cache in gem5 handles the coherency for IO memory accesses. 
Please let me know if my understanding is incorrect. 

Also, I changed the CPU type of the BigCluster to MinorCPU to keep both systems identical https://github.com/gem5/gem5/blob/48a40cf2f5182a82de360b7efa497d82e06b1631/configs/example/arm/fs_bigLITTLE.py#L85

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:
1. Make the above mentioned changes in devices.py and dist_bigLITTLE.py files.
2. Replace the util/dist/test/test-2nodes-AArch64.sh file with the attached file. In the attached file, I have enabled the interactive mode, caches and timing CPU model and provided the custom DTB file (which is also attached). 
3. Execute the test-2nodes-AArch64.sh to start the simulation
4. Once Linux bootup is complete, setup the ethernet in system 1 and 2 as below:
```
System 1:
/sbin/ifconfig eth0 hw ether 00:90:00:00:00:02
/sbin/ifconfig eth0 192.168.0.2 netmask 255.255.255.0 up

System 2:
/sbin/ifconfig eth0 hw ether 00:90:00:00:00:04
/sbin/ifconfig eth0 192.168.0.4 netmask 255.255.255.0 up
```
Now try to ping the system 2 from system 1:
`ping -c1 192.168.0.4`
When I execute this, simulation crash with the above-mentioned error. 

**Expected behavior**
Ping should happen without any error. 

**Host Operating System**
Ubuntu 22.04

**Host ISA**
 X86

**Compiler used**
gcc 11.3.0

**Additional information**
Linux kernel 5.15 was run on both systems with default gem5 config file. 

**test-2nodes-AArch64.sh  and dtb file that was used:**
[Archive.zip](https://github.com/gem5/gem5/files/12771011/Archive.zip)

Harshil2107, 2023-10-02:
@giactra and @andysan do you have any idea why this would be happening?

sammy17, 2023-10-05:
I tried to debug the issue but could not exactly figure out the root cause. 
Network communication works fine in atomic mode (mostly) without caches enabled. 
The crash only happens when the timing CPU models and caches are enabled. 
Below is a debug trace log with SMMU and Ethernet debug tracing enabled.  
Hope this helps to figure out the issue. 
Please let me know any suggestions to debug the issue further and resolve it. 
[crash.log](https://github.com/gem5/gem5/files/12812435/crash.log)

sammy17, 2023-10-17:
Hi @giactra, any update on this? 
I would really appreciate any help in resolving this issue. 
Also, are there any other devices that are verified to work with the SMMU in gem5?

sammy17, 2023-11-10:
I tried to debug this issue. As the error suggests, following assertion fails when performing an Ethernet packet transfer with SMMU enabled in the timing mode. 
```
// ReadEx and Writes require both invalidation and exlusivity, while reads
// require neither. Writebacks on the other hand require exclusivity but
// not the invalidation. Previously Writebacks did not generate upward
// snoops so this was never an issue. Now that Writebacks generate snoops
// we need a special case for Writebacks. Additionally cache maintenance
// operations can generate snoops as they clean and/or invalidate all
// caches down to the specified point of reference.
assert(cpkt->isWriteback() || cpkt->req->isUncacheable() ||
           (cpkt->isInvalidate() == cpkt->needsWritable()) ||
           cpkt->req->isCacheMaintenance());
```
Assert fail happen on the first WriteReq packet that arrive on lookupSnoop function. 
Following is the debug trace leading upto the crash:
```
7705626260000: system.realview.smmu: [t] requestor req  needs retry, qlen=0
7705626261000: system.realview.smmu.device_interfaces: [t] req from system.ethernet.dma addr=0xffe77b40 size=0x40
7705626261000: system.realview.smmu.device_interfaces.port: micro TLB hit vaddr=0xffe77ac0 amask=0xfffffffffffff000 sid=0 ssid=0 paddr=0x8269e000
7705626261000: system.realview.smmu.device_interfaces.port: IdHold: p=0x55ba913106e0 oid=9
7705626261000: system.realview.smmu.device_interfaces.port: IdRel:  p=0x55ba913106e0 oid=9
7705626262000: system.realview.smmu.device_interfaces: [t] req from system.ethernet.dma addr=0xffe77b80 size=0x40
7705626262000: system.realview.smmu: [t] requestor retr addr=0x8269ea40 size=0x40
7705626262000: system.membus: recvTimingReq: src system.membus.cpu_side_port[4] packet WriteReq [8269ea40:8269ea7f]
7705626262000: system.membus.snoop_filter: lookupRequest: src system.membus.cpu_side_port[4] packet WriteReq [8269ea40:8269ea7f]
7705626262000: system.membus.snoop_filter: lookupRequest:   SF value 0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010
7705626262000: system.membus: recvTimingReq: src system.membus.cpu_side_port[4] packet WriteReq [8269ea40:8269ea7f] SF size: 1 lat: 1
7705626262000: system.membus: forwardTiming for WriteReq [8269ea40:8269ea7f]
7705626262000: system.bigCluster.toL2Bus: recvTimingSnoopReq: src system.bigCluster.toL2Bus.mem_side_port[0] packet WriteReq [8269ea40:8269ea7f] ES
7705626262000: system.bigCluster.toL2Bus.snoop_filter: lookupSnoop: packet WriteReq [8269ea40:8269ea7f] ES
7705626262000: system.bigCluster.toL2Bus.snoop_filter: DEBUG: isWriteback: 0 isUncacheable: 0 isInvalidate: 0
7705626262000: system.bigCluster.toL2Bus.snoop_filter: DEBUG: needsWritable: 1 isCacheMaintenance: 0
gem5.opt: src/mem/snoop_filter.cc:237: std::pair<std::vector<gem5::QueuedResponsePort*>, gem5::Cycles> gem5::SnoopFilter::lookupSnoop(const gem5::Packet*): Assertion `cpkt->isWriteback() || cpkt->req->isUncacheable() || (cpkt->isInvalidate() == cpkt->needsWritable()) || cpkt->req->isCacheMaintenance()' failed.
```
I have printed the values that the asset checks as well. 
It seems like in the WriteReq packet, only the needsWritable attribute is set. 
From what I understood from the comment, isInvalidate attribute also should be set for this, is that correct? 
If I add the isInvalidate attribute to the WriteReq packet by changing it's definition at https://github.com/gem5/gem5/blob/e9da8d67bdbb23fbd0578a379f08f42bce50121d/src/mem/packet.cc#L79 it triggers other fundamental assertions in the abstract memory function. 

I traced back to where the WriteReq packet originate https://github.com/gem5/gem5/blob/e9da8d67bdbb23fbd0578a379f08f42bce50121d/src/dev/arm/smmu_v3_proc.cc#L122 But I am not sure how to fix the issue other than changing the packet type. 

**How can I solve this issue so that assertion does not fail? 
Can I change the packet type here?** 
It seems like WriteReq attributes contradict the comment in the assertion. 
Please let me know your thoughts. @giactra 
I would really appreciate any suggestions to solve this or debug this further.

giactra, 2023-11-10:
Hi @sammy17 , apologies for the late reply but I am being very busy in these days and I didn't find the time to reproduce the issue.

From an initial look, it seems to me the problem is that you are attaching the SMMU directly to the membus, which is a coherent bus and therefore requires coherent transactions. This is not the case for the MemCmd::WriteReq and that's why the snoop filter complains.

You should either attach the SMMU to the non coherent xbar (iobus) and rely on the IOCache to translate it into a coherent request, or you could place a cache in front of your SMMU. Let me know if this works, otherwise I will try to allocate some time next week to look into this more carefully.

sammy17, 2023-11-10:
Hi @giactra Thank you very much for the suggestions. 
I tried connecting the SMMU to the iobus as below. 
```
    # Following line is the only change, I added other lines for the completeness
    system.realview.attachSmmu([], system.iobus)
    # create NIC
    dev = IGbE_e1000()
    system.attach_pci(dev, dma_ports=[])
    system.ethernet = dev
    # Connect the device's DMA ports to the SMMU
    system.realview.smmu.connect(dev)
```
With this change, I ran the simulation with timing mode but during Linux bootup, the SMMU was not identified by the kernel properly. 
Linux bootup gave the following error for the SMMU:
```
[    0.032000] arm-smmu-v3 2b400000.smmuv3: unknown/unsupported TT endianness!
```
I checked the register that represent the SMMU endianess since the error suggests an issue with it. But the register is set correctly. 
And before when the SMMU is connected to the membus, Linux kernel recognized the SMMU without any issue. 
**Therefore, I believe this maybe related to a MMIO read issue when CPU reads the relevant SMMU register.** 
**Are there any other changes that I should be doing?** 
Please let me know your thoughts. 

Above message was the only message related to the SMMU in the kernel bootup, but I will add the full bootlog at the end of this message for your reference. 

I did not try the second solution you mentioned. Could you please let me know how I can connect a cache to the SMMU? 
Really appreciate any guidance on this.

sammy17, 2023-11-10:
Also, after changing the membus to iobus, when I tried to run the simulation in atomic CPU models (+ without caches), I got the following error. This issue was not seen when I run in timing mode as mentioned in the above connect, but I thought posting this as well incase it helps you to figure out the issue. 
```
src/dev/isa_fake.cc:59: warn: Device system.membus.badaddr_responder accessed by read to address 0x2b400000 size=4
src/cpu/simple/atomic.cc:405: panic: panic condition pkt.isError() occurred: Data fetch ([0x2b400000:0x2b400004]) failed: BadAddressError [2b400000:2b400003] UC
Memory Usage: 4910980 KBytes
Program aborted at tick 60713899500
```

It seems like there is an error when trying to access 0x2b400000 which is the SMMU base addr. 
Is this because caches are not available in atomic mode? 
Is the iocache enabled by default in timing mode or I have to enable it through an option?

giactra, 2023-11-11:
> Also, after changing the membus to iobus, when I tried to run the simulation in atomic CPU models (+ without caches), I got the following error. This issue was not seen when I run in timing mode as [...]

The fact that you get that error is logical; it’s weird that you don’t get it in timing mode…
Now that the SMMU is in the IOBus, its MMIO address range is not registered to the coherent xbar (membus). So when a request arrives, the membus doesn’t know where to dispatch it and that’s the error you are getting. You need to extend the iobridge ranges parameter to include the SMMU so that the membus will know that a request in that range should be forwarded to the bridge. 
Check the attachOnChipIO implementation to see how the bridge ranges get populated

sammy17, 2023-11-11:
Thanks for the quick response @giactra !
It finally worked!
I added the SMMU addr range to the iobridge in the system init function where the iobridge is initialized and the SMMU was properly initialized during Linux boot. 
I can communicate between two systems without the initial crash. 
Since the initial issue is fixed, I will close this issue. Thank you very much for your help!

Unfortunately, I am facing another, independent issue for which I just filed another issue - https://github.com/gem5/gem5/issues/548. 
Hope you can take a look and help me get that resolved as well. 
Thank you in advance!

closed 2023-11-11 by sammy17