gem5/gem5 #382: Adding Vector Segmented Loads/Stores to RISC-V V 1.0 implementation

ivanfv, 2023-10-02:
Hello,

I am opening this issue to ask if someone is working on the Vector Segmented Loads/Stores to the RISC-V Vector 1.0 implementation (e.g., vlseg, vsseg), which, if I am not wrong, are not supported by gem5 yet.

For the moment being, I am checking 1) how LD2, LD3 and LD4 Arm's instructions are implemented and 2) how vector register gather could help.

I will report the advances over here once I have significant progress.

Best,
Ivan

CC @adriaarmejach

ivanaamit, 2024-01-08:
Hi @ivanfv, I'm just checking to see the status of this issue. Are you currently working on it? Thanks.

ivanfv, 2024-01-09:
Hi @ivanaamit , yes, I am working on this. Currently, I am validating the vector segmented loads implementation and I plan to create a pull request about them very soon. Thanks for checking this!

ivanaamit, 2024-03-26:
Since #851 and #913 have been merged, I am closing this issue. @ivanfv, please reopen it if I am mistaken. Thanks.

closed 2024-03-26 by ivanaamit