gem5/gem5 #383: md5_utils.py uses internal _hashlib import

voertler, 2023-10-02:
**Describe the bug**
src/python/gem5/resources/md5_utils.py uses the internal _hashlib import. This can cause problems with custom Python versions which are built without OpenSSL support.

```
ModuleNotFoundError: No module named '_hashlib'

At:
  src/python/gem5/resources/md5_utils.py(29): <module>

```

Is the use of the internal HASH really neccessary?

BobbyRBruce, 2023-10-02:
@voertler : I'm unfamiliar with how this is implemented in Python. To me it's just an md5 module which comes as part of the Python Standard Library. To help me understand, Which of the following is true:

1. The use of `hashlib` and/or `_hashlib` in gem5 causes this `ModuleNotFound` exception. The only way to fix it is to remove both the `hashlib` and `_hashlib` dependencies.
2. `hashlib` is fine but importing from `_hashlib` causes problems. If we can remove the `_hashlib` dependency we can keep all the other hashing operations.

[Looking at the code](https://github.com/gem5/gem5/blob/48a40cf2f5182a82de360b7efa497d82e06b1631/src/python/gem5/resources/md5_utils.py) it seems to me like I could remove `from _hashlib import HASH as Hash` as this is only used to provide typing information. However, we do need to keep this general MD5 hashing functionality provided by `hashlib`. We use it to verify our downloading of resources are uncorrupted and correct. There are cases where we update a resource (say, a disk image) and the only way someone's local instance of gem5 knows it needs to redownload the resource is by checking the hash values. Perhaps this could have been engineered differently to avoid hashing  but it's quite heavily embedded into our system now so it's not something i'm comfortable trying to remove.

So, if the situation is (1.) then I'd say OpenSSL is just a requirement for us, unless you want to copy and paste the md5 algorithm into the gem5 codebase to avoid importing from the standard library. If the situation is (2.) then I'd welcome a PR to rectify this annoyance. It would be a harmless change.

PS: Out of curiosity, what is the case where Python is build without OpenSSL support?

BobbyRBruce, 2023-10-10:
Much appreciated for your work on this @voertler ! Closing this issue as fixed/completed.

closed 2023-10-10 by BobbyRBruce