gem5/gem5 #395: Feature request: Adding 9p support to the riscv-fs

Harshil2107, 2023-10-05:
I'm submitting this feature request in response to a message posted on the gem5-users mailing list, which can be found here: [link to the mail](https://www.mail-archive.com/gem5-users@gem5.org/msg21961.html). The user encountered difficulties while attempting to integrate 9p support into riscv-fs.

In Arm systems, 9p support is facilitated through the `def attach_9p()` function located at `configs/common/FSConfig.py`. Introducing this feature into RISC-V would be a valuable enhancement.

voertler, 2023-10-10:
Thanks for picking up this issue (I wrote the message on the mailing-list). I think this would be a great feature if it work for all platforms within the stdlib. @powerjg mentioned that also the device tree has to be modified, so I looked at the old FS implementation but I could not find anywhere how the device tree is modified when at 9p device is attached in the ARM platform. Any hint would be welcome.

This is the modification of the platform that I tried:
```
class COSRiscvBoard(RiscvBoard):
    @overrides(AbstractSystemBoard)
    def _setup_io_devices(self) -> None:
        super()._setup_io_devices()
        self.add_riscv_9p()

    def add_riscv_9p(self):
        print(f"Function _setup_board Adding 9p device to platform")
        self.viopci = PciVirtIO(pci_bus=0, pci_dev=1, # Hardcoded for the platform, Ethernet in RiscV Board is 0
                               pci_func=0, InterruptPin=1,
                               InterruptLine=1)
        self.viopci.vio = VirtIO9PDiod()
        viodir = os.path.realpath(os.path.join(m5.options.outdir, "9p"))
        self.viopci.vio.root = os.path.join(viodir, "share")
        self.viopci.vio.socketPath = os.path.join(viodir, "socket")
        os.makedirs(self.viopci.vio.root, exist_ok=True)
        if os.path.exists(self.viopci.vio.socketPath):
            os.remove(self.viopci.vio.socketPath)

        self.viopci.host = self.platform.pci_host
        self.viopci.pio = self.iobus.mem_side_ports
        self.viopci.dma = self.iobus.cpu_side_ports
```

powerjg, 2023-10-10:
The device tree code in Arm is *very* difficult to follow. It's hidden in about 12 different places, and none of them are consistent with others. I **think** the code you're looking for would be here: https://github.com/gem5/gem5/blob/stable/src/dev/arm/VirtIOMMIO.py#L57

What I would suggest is running a Arm simulation with 9p enabled and looking at the device tree file that's generated in `m5out`. Then, you can try to copy that into the function that creates the device tree for RISC-V. Given the spaghetti nature of this code, I wouldn't trust anything except for looking at the output.

(Note that instead of leveraging the "hidden" device tree code that Arm uses, I consolidated all of it to a single place in the RISC-V FS board so that it's easier to understand. This *could* result in some duplicated code (it hasn't yet), but it makes it much easier to modify/extend.)

abmerop, 2023-10-15:
I think 9p would be nice to have for *every* ISA actually. So maybe not RISC-V specific.  It's a very nice way to get files from the host system in full system without having to do disk image modifications. Maybe this issue could be made more general? 

For example, I implemented this in GPUFS on my own branch, however I am seeing a Linux error and not the python issues being discussed here

voertler, 2023-10-20:
@abmerop I would also support a general 9p device that works with every ISA, What kind of Linux error did occur? The challenge is to generate the correct dtb for 9p support within Linux.

abmerop, 2023-10-20:
> @abmerop I would also support a general 9p device that works with every ISA, What kind of Linux error did occur? The challenge is to generate the correct dtb for 9p support within Linux.

It's been a while so I had to retest this.  The specific "error" I am seeing from dmesg when running the mount command within gem5 is: 

`[  138.992647] 9pnet_virtio: no channels available for device gem5`

Searching for the error was not much help. I mostly copy and pasted everything from the ARM configs including the mount commands, but I seem to be missing something related to making a "gem5 device" known.  This is the full log:

```bash-5.0# dmesg -n8
bash-5.0# lspci -v
00:04.0 IDE interface: Intel Corporation 82371AB/EB/MB PIIX4 IDE (prog-if 80 [ISA Compatibility mode-only controller, supports bus mastering])
        Flags: bus master, 66MHz, user-definable features, fast devsel, latency 64
        Memory at 000001f0 (32-bit, non-prefetchable) [virtual] [size=8]
        Memory at 000003f0 (type 3, non-prefetchable) [virtual]
        Memory at 00000170 (32-bit, non-prefetchable) [virtual] [size=8]
        Memory at 00000370 (type 3, non-prefetchable) [virtual]
        I/O ports at 1020 [virtual] [size=16]
        Expansion ROM at c0080000 [virtual] [disabled] [size=2K]
        Kernel driver in use: ata_piix
        Kernel modules: pata_acpi

00:06.0 Unassigned class [ff00]: Red Hat, Inc. Virtio network device
        Subsystem: Red Hat, Inc. Virtio network device
        Flags: bus master, 66MHz, user-definable features, fast devsel, latency 64
        I/O ports at 1000 [size=32]
        Expansion ROM at c0080800 [virtual] [disabled] [size=2K]
        Kernel driver in use: virtio-pci

00:08.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 XTX [Radeon Vega Frontier Edition] (prog-if 00 [VGA controller])
        Flags: bus master, 66MHz, user-definable features, fast devsel, latency 0, IRQ 14
        Memory at 400000000 (64-bit, non-prefetchable) [size=16G]
        Memory at 100000000 (64-bit, non-prefetchable) [size=2M]
        Memory at c0000000 (32-bit, non-prefetchable) [size=512K]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [80] Express Endpoint, MSI 00
        Kernel driver in use: amdgpu
        Kernel modules: amdgpu

bash-5.0# mount -t 9p -o trans=virtio,version=9p2000.L,aname=/full/host/path/to/gem5/m5out/9p/share/,debug=0x01 gem5 /mnt
[  138.956644] FS-Cache: Loaded
[  138.980646] 9p: Installing v9fs 9p2000 file system support
[  138.984646] FS-Cache: Netfs '9p' registered for caching
[  138.992647] 9pnet_virtio: no channels available for device gem5
mount: /mnt: special device gem5 does not exist.
bash-5.0# ls /mnt
bash-5.0#```

Preliminary [code on my branch](https://github.com/abmerop/gem5/commit/3826a4a2f7c6bea7cf188b49f80d124bf098ef8d)

voertler, 2023-10-26:
A small update: After inspecting the device tree of the ARM platform, it seems that this dts entry is responsible for the 9p  device:
https://github.com/gem5/gem5/blob/e9da8d67bdbb23fbd0578a379f08f42bce50121d/system/arm/dt/platforms/vexpress_gem5_v1_base.dtsi#L134-L138
referenced by this device.
https://github.com/gem5/gem5/blob/e9da8d67bdbb23fbd0578a379f08f42bce50121d/src/dev/arm/RealView.py#L1426-L1428

According to the ARM FVP documentation this where their 9p device is added in the platform.
https://developer.arm.com/documentation/100966/1123/Getting-Started-with-Fixed-Virtual-Platforms/Virtio-P9-device-component?lang=en

So I tried adding the dts entry and a similar device in addition to the VirtIO9PDiod device in my RISC-V platform

`       
 self.p9mmvio= RiscvMmioVirtIO(
 pio_addr=0x1C140000, pio_size=0x1000, 
 interrupt_id=0x2b,
 )
 self._off_chip_devices.append(self.p9mmvio)
`

Still I get the same error when mounting: mount: mounting gem5 on /mnt/9p failed: No such device
@andysan and others Any: further Ideas? I've for example no idea, what the correct interrupt should be.

andysan, 2023-10-26:
The errors seen by @abmerop and @voertler are most likely caused by very different parts of the stack. I _think_ the error seen by @abmerop is probably caused by a misconfigured diod server. I has been a long time since I touched 9p and diod though, so I don't know for sure.

In general, if you have working PCI, there really isn't any reason to use an MMIO-based VirtIO interface. It's usually a waste of time and adds a lot of error-prone configuration steps (discovery and interrupts being the main ones). The only reason we include MMIO VirtIO in the Arm boards is to be compatible with the FVP platform.

The main problem with MMIO-based VirtIO is that gem5 lacks a platform-independent way to specify interrupts which results in platform-dependent implementations. There are some complexities in the way interrupts are described in device trees as well since each interrupt controller (at least in principle) could have its own way to specify interrupts. This is a problem for MMIO base addresses as well since some boards have weird ways of describing buses which means that the format of the `reg` field varies. These are all non-issues for PCI since each platform has its own PCI host device that handles all of the interrupt routing and PCI has its own discovery mechanism.

There are a few debug flags that you can enable to figure out what is going on here. The first step is to check if the VirtIO transport is working correctly. You should be able to enable the various VIO debug flags and check that the device sees requests from the host and that interrupts are delivered as expected. This seems to be working for @abmerop, but probably not for @voertler. I would strongly suggest trying this with a simple device like a UART before crossing into the land of dragons. Once you have established that the transport layer works, switch back to 9p and diod. At this point, you'll need to debug the diod configuration which is a bit more complex. I'm afraid I can't remember much about that though. My guess is that @abmerop hasn't specified the mount name correctly.