gem5/gem5 #403: Make cache recorder future proof and more elegant

v-ramadas, 2023-10-05:
The cache recorder is used during cache cooldown and warmup to either issue flushes to memory or populate cache. It uses a sequencer map to track all the L1 caches (https://github.com/gem5/gem5/blob/stable/src/mem/ruby/system/RubySystem.cc#L180) and calls their makeRequest() (https://github.com/gem5/gem5/blob/stable/src/mem/ruby/system/CacheRecorder.cc#L100) function during cooldown/warmup. Pull request #377 extends this support to GPU coalescers by adding a coalescer map to the cache recorder. It does so by duplicating code within the cache recorder to both populate the coalescer map and issue requests to the coalescer. If some other component requires flush support in the future, this approach would lead to more code duplication.

Would it be a good idea to maintain a list of RubyPort instances instead of separate lists for sequencers and gpu coalescers; i.e. use `std::vector<RubyPort>` instead of `std::vector<Sequencer>` at https://github.com/gem5/gem5/blob/stable/src/mem/ruby/system/RubySystem.cc#L180? Since makeRequest() is virtual in RubyPort and overriden by derived classes Sequencer and GPUCoalescer, the RubyPort map can be populated with pointers to either the sequencer or coalescer (or any similar component), and no code duplication is needed when issuing the makeRequest() call. Would this be a good (more elegant) alternative?

powerjg, 2023-10-05:
Sounds like a good idea to me!

mattsinc, 2023-10-16:
@v-ramadas should confirm, but since #448 was merged in, yes I believe this can be closed.

v-ramadas, 2023-10-16:
Yes, will close the issue.

closed 2023-10-16 by v-ramadas