gem5/gem5 #440: Add ".devcontainer" directory for running/developing gem5 in a container and via GitHub Codespaces

BobbyRBruce, 2023-10-11:
A introduction to dev containers can be found [here](https://docs.github.com/en/codespaces/setting-up-your-project-for-codespaces/adding-a-dev-container-configuration/introduction-to-dev-containers).

In short, they are "Development Containers"; Docker containers provided to create an environment to develop a specific software project in. For example a Dev Container typically contains all dependencies and tools needed to carry out code development. This lessens the burden on new developers means developers can, essentially, develop from any machine they wish.

In addition to this VSCode has great integration with DevContainer. Plugins and VSCode settings can be configured via a Dev Container. When a project containing a Dev Container specification is opened in VS code, VSCode will give the user the option to load vs code in the container, thus immediately jumping into a tailored development environment.

GitHub offers "[CodeSpaces](https://docs.github.com/en/codespaces/overview)" which allows for users, via the web-browser, to spin up Dev Containers for any repo they wish to develop for. In gem5 you _can_ click "CodeSpace" (an alternative under "Clone") on the main repo page but, at present, this creates GitHubs default development container which has none of the dependencies you require to develop for gem5. It is essentially useless.

It would be nice to have our own gem5 Dev Container setup for gem5.

Instructions on creating a dev container for a project can be found here: https://code.visualstudio.com/docs/devcontainers/create-dev-container.

BobbyRBruce, 2023-12-20:
Note: This should include the gem5 Slicc VS Code Extension: https://github.com/orgs/gem5/discussions/480

ivanaamit, 2024-04-15:
@BobbyRBruce, please re-open this issue if it was not fully implemented in #911.

closed 2024-04-15 by ivanaamit